svpcom/wfb-ng #136: [BUG]

maxipalay, 2021-05-26:
**Describe the bug**
Installation fails on ubuntu Focal (20.04).

**To Reproduce**
Steps to reproduce the behavior:
Installation steps until executing make deb.

**Screenshots**
removing 'wifibroadcast-trunk' (and everything under it)
dpkg-buildpackage: warning:     debian/changelog(l1): version 'trunk-1' is invalid: version number does not start with digit
LINE: wifibroadcast (trunk-1) unstable; urgency=low
dpkg-buildpackage: info: source package wifibroadcast
dpkg-buildpackage: info: source version unknown
dpkg-buildpackage: error: version number does not start with digit
Traceback (most recent call last):
  File "./setup.py", line 62, in <module>
    setup(
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/setuptools/__init__.py", line 162, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/bdist_deb.py", line 34, in run
    self.run_command('sdist_dsc')
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/sdist_dsc.py", line 137, in run
    build_dsc(debinfo,
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 1569, in build_dsc
    dpkg_buildpackage(*args, cwd=fullpath_repackaged_dirname)
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 585, in dpkg_buildpackage
    process_command(args, cwd=cwd)
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 226, in process_command
    check_call(args, cwd=cwd)
  File "/home/ubuntu/develop/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
make: *** [Makefile:53: deb] Error 1

**Your setup (please complete the following information):**
 - Ubuntu 20.04
 - Hardware: Rpi 4

maxipalay, 2021-05-26:
Seems the issue is in wifibroadcast/version.py

I don't have python2 installed and the env in version.py is /usr/bin/env python.

Changing it to /usr/bin/env python3 solved the issue.

Leaving here for documentation purposes.

closed 2021-05-26 by maxipalay

AlexNester-QuadX, 2023-11-01:
Have the same problem.
But I havn't any env python or env python3 in /usr/bin/

just only env but if I change to #!/usr/bin/env its no work
any installation like

sudo apt-get install python-is-python3

sudo apt install python-minimal

system
 Ubuntu 22.04.3 LTS
(ubuntu under Windows)

`sudo ./scripts/install_gs.sh wifi0
Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:3 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:5 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
debhelper is already the newest version (13.6ubuntu1).
iw is already the newest version (5.16-1build1).
libpcap-dev is already the newest version (1.10.1-4build1).
libsodium-dev is already the newest version (1.0.18-1build2).
python3-serial is already the newest version (3.5-1).
dh-python is already the newest version (5.20220403).
virtualenv is already the newest version (20.13.0+ds-2).
python3-all is already the newest version (3.10.6-1~22.04).
python3-future is already the newest version (0.18.2-5ubuntu0.1).
python3-pyroute2 is already the newest version (0.6.4-3ubuntu2).
python3-twisted is already the newest version (22.1.0-2ubuntu2.3).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
rm -rf deb_dist
/home/t0968639585/wfb-ng/env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
running bdist_deb
running sdist_dsc
running egg_info
writing wfb_ng.egg-info/PKG-INFO
writing dependency_links to wfb_ng.egg-info/dependency_links.txt
writing entry points to wfb_ng.egg-info/entry_points.txt
writing top-level names to wfb_ng.egg-info/top_level.txt
reading manifest file 'wfb_ng.egg-info/SOURCES.txt'
adding license file 'LICENSE.txt'
writing manifest file 'wfb_ng.egg-info/SOURCES.txt'
running sdist
running check
creating wfb-ng-23.8.14.60981
creating wfb-ng-23.8.14.60981/scripts
creating wfb-ng-23.8.14.60981/scripts/default
creating wfb-ng-23.8.14.60981/wfb_ng
creating wfb-ng-23.8.14.60981/wfb_ng.egg-info
creating wfb-ng-23.8.14.60981/wfb_ng/conf
copying files to wfb-ng-23.8.14.60981...
copying LICENSE.txt -> wfb-ng-23.8.14.60981
copying README.md -> wfb-ng-23.8.14.60981
copying setup.py -> wfb-ng-23.8.14.60981
[... 82 lines cut ...]
dpkg-buildpackage: error: fakeroot debian/rules clean subprocess returned exit status 1
Traceback (most recent call last):
  File "/home/t0968639585/wfb-ng/./setup.py", line 39, in <module>
    setup(
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/__init__.py", line 103, in setup
    return distutils.core.setup(**attrs)
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
    super().run_command(command)
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/stdeb/command/bdist_deb.py", line 67, in run
    util.process_command(syscmd, cwd=target_dirs[0])
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 226, in process_command
    check_call(args, cwd=cwd)
  File "/home/t0968639585/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 1
make: *** [Makefile:50: deb] Error 1
t0968639585@DESKTOP-V36GQNC:~/wfb-ng$`