svpcom/wfb-ng #167: [BUG] latency_Test.py - mixing sessions

pasdVn, 2021-09-29:
**Describe the bug**
I tried to use {{latency_test.py}} for testing the performance of different wifi devices. For some reason I nearly always have the case, that some of the messages of one test are (obviously) received only in the next test so that I always get those "bad session" log messages.
I tired to understand/debug the problem on my own but I can't find the reason.

**To Reproduce**
Start sender receiver:
```

DEV_RX=wlx00c0ca98eb12
DEV_TX=wlx00c0ca98ebd1

PARAM_N=12
PARAM_K=8
/usr/bin/wfb_tx -p 1 -u 14601 -K ${SCRIPT_DIR}/drone_test.key -B 20 -G long -S 0 -L 0 -M 1 -k $PARAM_K -n $PARAM_N -T 0 $DEV_TX &
/usr/bin/wfb_rx -p 1 -u 14600 -K ${SCRIPT_DIR}/gs_test.key -k $PARAM_K -n $PARAM_N $DEV_RX
```

Then I am running the tests (with some log example):
```
python3 latency_test.py  14601 14600 140 100

2021-09-29 20:47:23+0100 [-] Log opened.
2021-09-29 20:47:23+0100 [-] Session: 27269
2021-09-29 20:47:23+0100 [-] PacketSource starting on 37622
2021-09-29 20:47:23+0100 [-] PacketSink starting on 14600
2021-09-29 20:47:26+0100 [-] (UDP Port 37622 Closed)
2021-09-29 20:47:28+0100 [-] Sent 150, Lost 10, Dup: 0, Packet rate: 50 pkt/s, Bitrate: 0.06 MBit/s, Lmin 1.59 ms, Lmax 243.35 ms, Lavg 50.40 ms
2021-09-29 20:47:28+0100 [-] (UDP Port 14600 Closed)
2021-09-29 20:47:28+0100 [-] Session: 10452
2021-09-29 20:47:28+0100 [-] PacketSource starting on 48290
2021-09-29 20:47:28+0100 [-] PacketSink starting on 14600
2021-09-29 20:47:31+0100 [-] (UDP Port 48290 Closed)
2021-09-29 20:47:33+0100 [-] Sent 75, Lost 9, Dup: 0, Packet rate: 25 pkt/s, Bitrate: 0.03 MBit/s, Lmin 1.68 ms, Lmax 482.83 ms, Lavg 96.24 ms
2021-09-29 20:47:33+0100 [-] (UDP Port 14600 Closed)
2021-09-29 20:47:33+0100 [-] Session: 42351
2021-09-29 20:47:33+0100 [-] PacketSource starting on 41888
2021-09-29 20:47:33+0100 [-] PacketSink starting on 14600
2021-09-29 20:47:33+0100 [PacketSink (UDP)] bad session 10452 != 42351 #69, already got 0 packets
2021-09-29 20:47:33+0100 [PacketSink (UDP)] bad session 10452 != 42351 #70, already got 0 packets
2021-09-29 20:47:33+0100 [PacketSink (UDP)] bad session 10452 != 42351 #71, already got 0 packets
2021-09-29 20:47:33+0100 [PacketSink (UDP)] bad session 10452 != 42351 #72, already got 0 packets
2021-09-29 20:47:33+0100 [PacketSink (UDP)] bad session 10452 != 42351 #74, already got 0 packets
2021-09-29 20:47:36+0100 [-] (UDP Port 41888 Closed)
2021-09-29 20:47:38+0100 [-] Sent 39, Lost 2, Dup: 0, Packet rate: 13 pkt/s, Bitrate: 0.01 MBit/s, Lmin 1.77 ms, Lmax 540.37 ms, Lavg 108.07 ms

```

As you can see some messages with the session id of the first test (10452) are - for some reason - only received in the second test (while some messages are lost in the first test).

I am testing in a very noisy environment, but I think this should no happen anyways (I also tried to increase the wait time etc).

I even sometimes have the case, that I run latency_test.py several times (even with some seconds/minutes delay in between) an I get the session ID of the last execution (last test) in the next run. It even seems, as If the messages send by wfb_rx are somehow "queued" until some new udp socket is create and bind to it's destination port (which is completely nonsense - or I do not understand udp correctly).
One log as prove:

```
2021-09-29 20:47:53+0100 [-] Session: 16558
2021-09-29 20:47:53+0100 [-] PacketSource starting on 40916
2021-09-29 20:47:53+0100 [-] PacketSink starting on 14600
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 49519 != 16558 #9, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 49519 != 16558 #11, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #0, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #1, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #2, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #3, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #4, already got 0 packets
2021-09-29 20:47:53+0100 [PacketSink (UDP)] bad session 55579 != 16558 #5, already got 0 packets
^C2021-09-29 20:47:55+0100 [-] Received SIGINT, shutting down.
2021-09-29 20:47:55+0100 [PacketSource (UDP)] (UDP Port 40916 Closed)
2021-09-29 20:47:55+0100 [PacketSink (UDP)] (UDP Port 14600 Closed)
2021-09-29 20:47:55+0100 [-] Main loop terminated.
2021-09-29 20:47:55+0100 [-] Exiting with code 0

< did some stuff here in between>

pi@gcs:~/Entwicklung/wifibroadcast/telemetry $ 
pi@gcs:~/Entwicklung/wifibroadcast/telemetry $ python3 latency_test.py  14601 14600 140 100
2021-09-29 21:04:29+0100 [-] Log opened.
2021-09-29 21:04:29+0100 [-] Session: 24177
2021-09-29 21:04:29+0100 [-] PacketSource starting on 45784
2021-09-29 21:04:29+0100 [-] PacketSink starting on 14600
2021-09-29 21:04:29+0100 [PacketSink (UDP)] bad session 16558 != 24177 #3, already got 0 packets
2021-09-29 21:04:29+0100 [PacketSink (UDP)] bad session 16558 != 24177 #4, already got 0 packets
2021-09-29 21:04:29+0100 [PacketSink (UDP)] bad session 16558 != 24177 #5, already got 0 packets
2021-09-29 21:04:29+0100 [PacketSink (UDP)] bad session 16558 != 24177 #6, already got 0 packets
2021-09-29 21:04:29+0100 [PacketSink (UDP)] bad session 16558 != 24177 #7, already got 0 packets
< and so on>
```

**Expected behavior**
This should not happen - all messages of one test should also get received within this test.

**Your setup (please complete the following information):**
 - raspbian buster
 - current master branch of wifibroadcast
 - 2 AWUS036NHA 

**Confirm you read**
- [ x]  https://github.com/svpcom/wifibroadcast/blob/master/README.md
- [ x]  https://github.com/svpcom/wifibroadcast/wiki/Setup-HOWTO

pasdVn, 2021-09-29:
One more note: I needed to tweak latency_test.py a bit  to be able to run with python3 - but I experienced the same behaviour with python2.

svpcom, 2021-09-30:
It is normal behavior due to FEC operations. FEC(N, K) use K input packets and issue (N - K) checksums. FEC can encode and decode only full block (K messages). If number of packets to send is not divisible by K, then TX FEC block will not closed, and only new latency test run will close it. In case of packet loss you will wait for K packets of the last FEC block on the RX side which will only arrive when new test run close FEC block (and issue checksum packets) on TX.

There are two ways to deal with it:
1. Restart wfb_rx and wfb_tx between latency test runs
2. Add `-T timeout_in_ms` to wfb_tx to close FEC block on timeout (it will issue null packets to FEC if no new packets arrives during specified timeout). For example you can add `-T 100`

pasdVn, 2021-10-01:
Thank you a lot for this explanation - that was exactly what I was missing!

closed 2021-10-01 by pasdVn