svpcom/wfb-ng #214: MavSysID and MavConpID

svpcom, 2022-04-14:
### Discussed in https://github.com/svpcom/wifibroadcast/discussions/213

<div type='discussions-op-text'>

<sup>Originally posted by **Dronotique** April 14, 2022</sup>
Hi all,

For the moment, WFB send a RSSI status Mavlink message with a static Vehicule Sys ID and a static component id hard coded line 65 in this file https://github.com/svpcom/wifibroadcast/blob/b610c5030019f41e9b04ea967abfbf6a39cbd709/telemetry/proxy.py

Following the mavlink documentation (https://mavlink.io/en/messages/common.html#MAV_COMP_ID_TELEMETRY_RADIO) I think the component ID 68 should be a better choice.

By the way, it would be great if we can configure this values (compid AND sysid) in the configuration file!
</div>

svpcom, 2022-04-14:
Done. Please check

Dronotique, 2022-04-15:
I tried since this morning to make it works without success.
I don't understand why!

I deleted all old files (I think so!) and reinstall the new version on each GS and drone and I still have a sysID=3 and compId=242

The drone conf
`
[drone_mavlink]
peer = 'listen://0.0.0.0:14550'   # incoming connection
bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index
mavlink_sys_id = 1    # for injected rssi packets
mavlink_comp_id = 68  # MAV_COMP_ID_TELEMETRY_RADIO
`

The GS conf
`
[gs_mavlink]
keypair = 'gs.key'
peer = 'connect://127.0.0.1:14650'  # outgoing connection
bandwidth = 20     # bandwidth 20 or 40 MHz
short_gi = False   # use short GI or not
stbc = 1           # stbc streams: 1, 2, 3 or 0 if unused
mcs_index = 1      # mcs index
mavlink_sys_id = 1    # for injected rssi packets
mavlink_comp_id = 68  # MAV_COMP_ID_TELEMETRY_RADIO
`

svpcom, 2022-04-15:
Do you use `git clone` for master branch or tgz from `Releases` tab?
How you install WFB? Use `pip install` or `make deb` and `dpkg -i wifibroadcast*.deb`?

svpcom, 2022-04-27:
@Dronotique Any success with mavlink_sys_id/mavlink_comp_id testing?

Dronotique, 2022-04-27:
I re-tested without success in the begin of th week.
I'll try again on friday and i'll get you in touch.

Dronotique, 2022-05-09:
Hello,

I finally managed to make it works. I had to search and delete all old files (python cache) and now it's OK.

Thanks a lot!

svpcom, 2022-05-09:
fixed

closed 2022-05-09 by svpcom