svpcom/wfb-ng #217: dpkg: error: cannot access archive 'wifibroadcast*.deb': No such file or directory

mmucahitkaya, 2022-05-15:
Hi, When i try to install wifibroadcast, i got this error.

dpkg: error: cannot access archive 'wifibroadcast*.deb': No such file or directory

[image]

svpcom, 2022-05-15:
Check that `make deb` finished without errors

mmucahitkaya, 2022-05-15:
I can't to fix this error. Here is screenshot for `sudo make deb`

[image]

svpcom, 2022-05-15:
`apt install python3-all dh-python debhelper`

mmucahitkaya, 2022-05-15:
Oh got, it's done. Thank you very much 🙂

[image]

closed 2022-05-15 by mmucahitkaya