svpcom/wfb-ng #246: Error building deb file

Jcwscience, 2022-09-01:
Jetson AGX Xavier, ubuntu 20.04 aarch64

when running make deb I receive this error

`
make[2]: Entering directory '/home/john/src/wifibroadcast/deb_dist/wfb-ng-22.8.11.29936'
python3 setup.py install --force --root=debian/wfb-ng --no-compile -O0 --install-layout=deb --prefix=/usr 
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help
error: option --install-layout not recognized
`

svpcom, 2022-09-01:
Try to replace `--upgrade pip==20.2.3 setuptools==44.1.1` with `--upgrade pip setuptools` in the Makefile, the run `make env` and `make deb`

Jcwscience, 2022-09-01:
One step further! Now I have this problem

`Traceback (most recent call last):
  File "./setup.py", line 62, in <module>
    setup(
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 973, in run_commands
    self.run_command(cmd)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/bdist_deb.py", line 34, in run
    self.run_command('sdist_dsc')
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/sdist_dsc.py", line 30, in run
    debinfo = self.get_debinfo()
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/stdeb/command/common.py", line 197, in get_debinfo
    debinfo = DebianInfo(
  File "/home/john/src/wifibroadcast/env/lib/python3.8/site-packages/stdeb/util.py", line 934, in __init__
    for line in long_description.split('\n'):
AttributeError: 'NoneType' object has no attribute 'split' `

svpcom, 2022-09-02:
``` .diff
diff --git a/setup.py b/setup.py
index 0ea3623..d105b51 100644
--- a/setup.py
+++ b/setup.py
@@ -80,5 +80,7 @@ setup(
     author="Vasily Evseenko",
     author_email="svpcom@p2ptech.org",
     description="The next generation of long-range packet radio link based on raw WiFi radio",
+    long_description=open('README.md').read(),
+    long_description_content_type='text/markdown',
     license="GPLv3",
 )
```

Jcwscience, 2022-09-02:
Ok that problem is gone but I’m back to the first issue again —install-layout not recognized

Thanks for the help by the way

svpcom, 2022-09-02:
What python version(s) do you have installed?
https://github.com/astraw/stdeb/issues/175

Jcwscience, 2022-09-02:
3.8.10

Jcwscience, 2022-09-02:
3.8.10

svpcom, 2022-09-02:
Try to:
```
apt install python3-stdeb
python3 ./setup.py --command-packages=stdeb.command bdist_deb
```

Jcwscience, 2022-09-02:
I’m going to have to try this in a while when I get back to my computer. I’ll let you know

Jcwscience, 2022-09-02:
Getting closer

`dpkg-source: error: version number does not start with digit`

svpcom, 2022-09-02:
Do you use `git clone` or tar.gz from releases page?

Jcwscience, 2022-09-02:
Got clone - checkout release

closed 2023-01-30 by svpcom