svpcom/wfb-ng #264: New feature: sending still images

maciek252, 2022-12-15:
I'm highly interested in the possibility of sending still images (photos) instead of the video stream. I guess it would need way less bandwidth and be more resistant to temporary network glitches. A photo could be updated at constant periods, say every 10 seconds, and during this time the tx could send the same photo over and over again (or its fragments if it has received an acknowledgment about some part successfully transmitted).

Motivation: I'm mostly flying by waypoints, so this could be a way of checking if everything goes fine, no unexpected masts, power lines etc. The system could switch to video if the available bandwidth would be appropriate.

JesterHal, 2022-12-15:
This is a very bad idea and makes the flight unsafe. You should always see in real-time where your going and "Return To Home" when you lose video feed. We discourage unsafe flight methods that are based only on GPS navigation. We have seen many times the GPS reporting false data because of glitches. So please fly safe and always remember that safety comes first.

maciek252, 2022-12-15:
I didn't say that I use only GPS navigation. I'm also using the lidar with a terrain map programmed into the companion computer, so I'm not relaying only on a GPS. If a discrepancy is detected, the drone enters the pre-programmed obstacle-avoiding route. Also, my flights are relatively short, I never fly say 20 km which is ubiquitous in the reports about the video links. In my opinion, this does not provide safety wrt to the collisions with other aircraft even when the video link works perfectly. I fly rather very different missions, say short-distance low altitude terrain-following flights around a mountain and into valleys.

JesterHal, 2022-12-15:
In my humble opinion no one else is going to use this feature. Everyone else prefers to have stable and robust video link, not still images (1 frame per 10 seconds). You could have a separate link for that through the tunnel link while streaming the regular video. But that's a different topic/request.

maciek252, 2022-12-15:
Maybe not many users are interested in such an option as the preferred mode, but it would be an alternative for the situations where there is not enough bandwidth for the correct video transmission. And this happens quite often.

svpcom, 2022-12-15:
@maciek252 wfb-ng is just a transport. It doesn't tied to any video format. Use `rtpjpegpay`  gstreamer element to send still images if you want