svpcom/wfb-ng #280: [FEATURE] good to have a better indication of video issues

maxlapshin, 2023-02-07:
For some unknown reasons my raspberry pi has lost his video module:

```
root@raspberrypi:/home/pi# ls -l /dev/video0
ls: cannot access '/dev/video0': No such file or directory
```

I think that it may be rather convenient to pass somehow indication of this issue to the QGC via mavlink.

Question is: what is the best way to pass it? Text message or some special command?

svpcom, 2023-03-16:
It seems that is autopilot dependent, but I'll be prefer to get periodical text notification into QGC console. Need to add some monitoring subsystem which will check for drone resources and issue mavlink messages if something bad happens. It could be standalone program which connect to localhost:5760 (wfb-ng internal mavlink proxy)

maxlapshin, 2023-03-17:
Got it.

Will return later with better proposal if I find some more generic and standard way of diagnostic this.

closed 2023-03-17 by maxlapshin