svpcom/wfb-ng #283: Does it support h265?

zhuwei, 2023-02-27:
Openipc setting H265 at the drone end
Is the ground of wfb-ng supported? How to set?

svpcom, 2023-02-27:
Hi,
openipc and wfb-ng using incompatible network protocols. You need to setup either wfb-ng or openhd on both ends.

closed 2023-02-27 by svpcom

zhuwei, 2023-02-27:
I can display the picture normally with h264
Openipc's fpv uses wfb

svpcom, 2023-02-27:
h265 is only supported on x86. RPI unable to decode h265 (only h264). To decode h265 on x86 use https://github.com/svpcom/wfb-ng-osd

zhuwei, 2023-02-27:
I haven't displayed images on RPI, I display them on my mobile phone by QGC

svpcom, 2023-02-27:
It should work out of box (QGC settings : RTP, H265). But openipc setup is not officially supported by wfb-ng, so ask on openipc forums