svpcom/wfb-ng #292: [BUG] don't compile in arm32v7/ubuntu

maxlapshin, 2023-03-16:
**Describe the bug**
`make deb` breaks inside debhelper

**To Reproduce**

my Dockerfile:

```
FROM arm32v7/ubuntu:latest

RUN apt update && \
  apt install -y build-essential git wget \
    virtualenv fakeroot debhelper \
    libpcap-dev libsodium-dev \
    python3-twisted python3-pyroute2 \
    python3-future python3-all

RUN git clone https://github.com/svpcom/wfb-ng.git --depth 1 --branch wfb-ng-23.01
WORKDIR /wfb-ng

RUN make deb
```

I launch it on arm64 ubuntu:

```
docker build -t wfb-pkg .
Sending build context to Docker daemon  3.072kB
Step 1/5 : FROM arm32v7/ubuntu:latest
 ---> f3a711451e14
Step 2/5 : RUN apt update &&   apt install -y build-essential git wget     virtualenv fakeroot debhelper     libpcap-dev libsodium-dev     python3-twisted python3-pyroute2     python3-future python3-all
 ---> Using cache
 ---> 60eff4e3cc58
Step 3/5 : RUN git clone https://github.com/svpcom/wfb-ng.git --depth 1 --branch wfb-ng-23.01
 ---> Using cache
 ---> 0be485ecd2c7
Step 4/5 : WORKDIR /wfb-ng
 ---> Using cache
 ---> 1b1e6159cb53
Step 5/5 : RUN make deb
 ---> [Warning] The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in 898c46793f8b
g++  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
src/radiotap.c: In function 'ieee80211_radiotap_iterator_init':
src/radiotap.c:116:34: warning: taking address of packed member of 'struct ieee80211_radiotap_header' may result in an unaligned pointer value [-Waddress-of-packed-member]
  116 |         iterator->_next_bitmap = &radiotap_header->it_present;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
g++  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
cc  -Wall -O2 -DWFB_VERSION='"23.1.16.78051-4886033f"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lpcap -lsodium
virtualenv env --python=/usr/bin/python3
created virtual environment CPython3.10.6.final.0-32 in 1730ms
  creator CPython3Posix(dest=/wfb-ng/env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==22.0.2, setuptools==59.6.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
./env/bin/pip install --upgrade pip setuptools stdeb
Requirement already satisfied: pip in ./env/lib/python3.10/site-packages (22.0.2)
Collecting pip
  Downloading pip-23.0.1-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 4.5 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./env/lib/python3.10/site-packages (59.6.0)
Collecting setuptools
  Downloading setuptools-67.6.0-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 8.2 MB/s eta 0:00:00
[... 115 lines cut ...]
  File "/wfb-ng/env/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/wfb-ng/env/lib/python3.10/site-packages/stdeb/command/sdist_dsc.py", line 137, in run
    build_dsc(debinfo,
  File "/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 1569, in build_dsc
    dpkg_buildpackage(*args, cwd=fullpath_repackaged_dirname)
  File "/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 585, in dpkg_buildpackage
    process_command(args, cwd=cwd)
  File "/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 226, in process_command
    check_call(args, cwd=cwd)
  File "/wfb-ng/env/lib/python3.10/site-packages/stdeb/util.py", line 59, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
make: *** [Makefile:48: deb] Error 1
The command '/bin/sh -c make deb' returned a non-zero code: 2
make: *** [Makefile:2: all] Error 2
```

**Expected behavior**

Should finish with building deb package

**Confirm you read**
- [x]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [x]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

maxlapshin, 2023-03-16:
https://askubuntu.com/questions/1263305/launchpad-builderror-cant-locate-debian-debhelper-sequence-python3-pm

> Add dh-python to Build-Depends in the debian/control file.

I suppose, this is useless

maxlapshin, 2023-03-16:
https://github.com/svpcom/wfb-ng/issues/62  should I stick to python2 instead of python3?

svpcom, 2023-03-16:
You need to install `dh-python`:
``` .sh
$ dpkg -S /usr/share/perl5/Debian/Debhelper/Sequence/python3.pm
dh-python: /usr/share/perl5/Debian/Debhelper/Sequence/python3.pm
```

svpcom, 2023-03-16:
@maxlapshin Now it should build deb files for any arch which supported in docker

closed 2023-03-16 by svpcom