svpcom/wfb-ng #306: Build for nanopi NEO-Core

Alex617923, 2023-09-05:
Hello, i want to build air and ground part wfb-ng for nanopi NEO-Core, after building rtl8812au drivers, cloning wfb-ng and installing all packages i try make deb and get

root@NanoPi-NEO-Core:~/wfb-ng# make deb
g++  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
cc  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
g++  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
src/tx.cpp: In function 'void data_source(std::shared_ptr<Transmitter>&, std::vector<int>&, int, bool)':
src/tx.cpp:375:119: warning: format '%llu' expects argument of type 'long long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
                         fprintf(stderr, "Incoming packet size > %" PRIu64 " and will be truncated\n", MAX_PAYLOAD_SIZE);
                                                                                                                       ^
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
cc  -Wall -O2 -DWFB_VERSION='"23.8.14.60981-ecdca20b"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lpcap -lsodium
virtualenv --python=/usr/bin/python3 /root/wfb-ng/env
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /root/wfb-ng/env/bin/python3
Also creating executable in /root/wfb-ng/env/bin/python
Installing setuptools < 51.3, pkg_resources, pip < 21, wheel...
  Complete output from command /root/wfb-ng/env/bin/python3 - "setuptools < 51.3" pkg_resources "pip < 21" wheel:
  Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
  File "/usr/share/python-wheels/pip-8.1.1-py2.py3-none-any.whl/pip/__init__.py", line 215, in main
  File "/root/wfb-ng/env/lib/python3.5/locale.py", line 594, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
----------------------------------------
...Installing setuptools < 51.3, pkg_resources, pip < 21, wheel...done.
Traceback (most recent call last):
  File "/usr/bin/virtualenv", line 9, in <module>
    load_entry_point('virtualenv==15.0.1', 'console_scripts', 'virtualenv')()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 719, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 991, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 918, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 812, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /root/wfb-ng/env/bin/python3 - "setuptools < 51.3" pkg_resources "pip < 21" wheel failed with error code 1
Makefile:17: recipe for target '/root/wfb-ng/env' failed
make: *** [/root/wfb-ng/env] Error 1

Can you help me?

svpcom, 2023-09-05:
`export LC_ALL=C LANG=C`
before build

Alex617923, 2023-09-08:
Thanks, i use newest armbian version and build success.

closed 2023-09-08 by svpcom