svpcom/wfb-ng #314: Speed up mavlink?

kellyschrock, 2024-01-04:
Hi, I've been using WFB with good success on a couple of drones I built. I modified the ground-station side to provide an access point so I can connect an Android tablet and my laptop, and a udp splitter that allows both of them to see video. The issue I'm having is that the mavlink connection seems really slow. If I attempt to upload a mission from an app running on my tablet or laptop, it takes a really long time to upload and fails pretty often from timeout errors. Is there a configuration change I can make to maybe give mavlink more bandwidth? Even if this comes at the expense of video quality, I'd be in favor of it. Thanks!

svpcom, 2024-01-06:
Could you provide your:
1. wifibrocast.cfg (from gs and drone) and exact version of wfb-ng used
2. screenshots of both `wfb-cli gs` and `wfb-cli drone` when you uploading a new mission to the drone

kellyschrock, 2024-01-06:
Hi, sure thing, I'm using 23.08. Files are attached.

FWIW, it's not that it's not working at all, but I was wondering if there's a way to control how much bandwidth each part of the connection (video, mavlink, tunnel) gets. I could in theory make mission uploads faster that way. Thanks.
[wfbcfg.zip](https://github.com/svpcom/wfb-ng/files/1385
[image]
0462/wfbcfg.zip)

svpcom, 2024-01-06:
```
peer = 'listen://0.0.0.0:14550'

# Uncoment if you need to forward mavlink stream
# peer = 'serial:/dev/ttyAMA'
#peer = 'serial:/dev/ttyS0:1500000' # Serial port for autopilot
```

do you use mavlink-router on the drone?

svpcom, 2024-01-06:
It seems that your speed is limited by serial port settings and/or mavlink-router

kellyschrock, 2024-01-06:
I actually use `cmavnode` for mavlink routing, and the serial port setting is 460800 baud. I'm not sure how I settled on that number, but I'll see if I can use a higher setting. That's likely to help.

kellyschrock, 2024-01-06:
I upped the baud rate to 921600 and that helped a bit.

closed 2025-05-06 by kellyschrock