svpcom/wfb-ng #332: wfb-cli gs cannot be used

mzhixian, 2024-04-18:
I deployed wfb-ng on tspi(RK3566) with kernel version 5.10.198. Enter wfb-cli gs and the following information is displayed：
[image]
[image]
Enter wfb_rx -K /etc/gs.key -c 127.0.0.1 -u 5600 -p 0 -i 7669206 wlxfca386b37542 and the following information is displayed：
[image]
This is dmesg：
[image]
What should I do？

svpcom, 2024-04-18:
`sudo journalctl -xu wifibroadcast@gs > gs.log`
`sudo journalctl -xu wifibroadcast@drone > drone.log`

svpcom, 2024-04-18:
`sudo ethtool -i wlxfca386b37542`

mzhixian, 2024-04-18:
> `sudo ethtool -i wlxfca386b37542`

[image]

mzhixian, 2024-04-18:
> `sudo journalctl -xu wifibroadcast@gs > gs.log` `sudo journalctl -xu wifibroadcast@drone > drone.log`
[gs.txt](https://github.com/svpcom/wfb-ng/files/15027454/gs.txt)

svpcom, 2024-04-18:
```
Apr 19 02:01:24 neons wfb-server[1685]: [common.abort_on_crash #info] Stopping reactor due to fatal error: [Errno 2] No such file or directory: b'/dev/net/tun'
```

Need to install tuntap driver (tun.ko)

mzhixian, 2024-04-19:
> sudo journalctl -xu wifibroadcast@gs > gs.log
It's working now but pkg don't increase.This is gs.log:
[image]
What should i do with this error :
Apr 19 02:01:24 neons wfb-server[1685]: [common.abort_on_crash #info] Stopping reactor due to fatal error: [Errno 2] No such file or directory: b'/dev/net/tun'

mzhixian, 2024-04-19:
> ```
> Apr 19 02:01:24 neons wfb-server[1685]: [common.abort_on_crash #info] Stopping reactor due to fatal error: [Errno 2] No such file or directory: b'/dev/net/tun'
> ```
> 
> Need to install tuntap driver (tun.ko)

It's working now.But PKG don't increase.This is gs.log:
[gs.txt](https://github.com/svpcom/wfb-ng/files/15040407/gs.txt)

svpcom, 2024-04-19:
```
pr 19 21:03:07 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] video rx: Unable to decrypt packet #0x233704
Apr 19 21:03:07 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] video rx: Unable to decrypt packet #0x233705
Apr 19 21:03:07 neons wfb-server[1644]: [server.AntennaProtocol.lineReceived #info] New session detected [video rx]: FEC=VDM_RS K=8, N=12, epoch=0
Apr 19 21:03:11 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] video rx: 178 packets lost
Apr 19 21:03:19 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] video rx: Error: socket error!
Apr 19 21:03:19 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] tunnel rx: Error: socket error!
Apr 19 21:03:19 neons wfb-server[1644]: [server.DbgProtocol.lineReceived #info] mavlink rx: Error: socket error!
Apr 19 21:03:19 neons wfb-server[1644]: [server.RXProtocol.processEnded #info] Stopped RX tunnel rx with code 1
Apr 19 21:03:20 neons wfb-server[1644]: [common.abort_on_crash #info] Stopping reactor due to fatal error: A process has ended with a probable error condition: process ended with exit code 1.
```

something bad with wifi card or there are: wpa_supplicant or dhclient which try to manage card. Check all steps from SetupHOWTO

svpcom, 2024-04-19:
Check `dmesg` that card didn't disconnect during wfb-ng running

mzhixian, 2024-04-22:
> Check `dmesg` that card didn't disconnect during wfb-ng running

I try all the SetupHOWTO，but the situation remains.I don't know what is the problem that makes my network card only work for a while.
[image]
[image]
[dmesg.txt](https://github.com/svpcom/wfb-ng/files/15064080/dmesg.txt)
[Uploading gs.txt…]()
[image]

mzhixian, 2024-04-22:
> gs.txt is missing

[gs.txt](https://github.com/svpcom/wfb-ng/files/15064407/gs.txt)