svpcom/wfb-ng #333: [Jetson Orin] wfb-ng/env/bin/pip: No such file or directory

lida2003, 2024-04-19:
Hi, following build commands as follows:

```
git clone -b stable https://github.com/svpcom/wfb-ng.git
cd wfb-ng
sudo ./scripts/install_gs.sh wifi0
```

And I got this error"make: /home/daniel/wfb/wfb-ng/env/bin/python: No such file or directory" on Jetson Orin.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
$ uname -a
Linux daniel-nvidia 5.15.122-tegra #1 SMP PREEMPT Mon Dec 18 21:24:25 PST 2023 aarch64 aarch64 aarch64 GNU/Linux
```

```
$ sudo ./scripts/install_gs.sh wlan1
Hit:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease
Hit:2 https://download.docker.com/linux/ubuntu jammy InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:8 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Hit:6 https://repo.download.nvidia.cn/jetson/common r36.2 InRelease
Hit:7 https://repo.download.nvidia.cn/jetson/t234 r36.2 InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  busybox-initramfs cpio initramfs-tools initramfs-tools-bin initramfs-tools-core klibc-utils libklibc linux-base linux-headers-5.15.0-102 linux-headers-5.15.0-102-generic usb-modeswitch
  usb-modeswitch-data zstd
Use 'sudo apt autoremove' to remove them.
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  vlc-plugin-qt libvlc5 libpostproc-dev vlc-data libvlccore9 vlc vlc-bin
  libiperf0 libjs-jquery-ui vlc-l10n libavdevice58 libopenexr25 python3-scipy
  libpostproc55 vlc-plugin-samba libswscale-dev libavdevice-dev libavcodec58
  iperf3 vlc-plugin-notify libavutil56 libswscale5 vlc-plugin-access-extra
  libavutil-dev vlc-plugin-skins2 vlc-plugin-video-splitter libswresample3
  vlc-plugin-video-output libavfilter-dev libavformat58 libpmix-dev
  libavformat-dev libavcodec-dev libpmix2 libvlc-bin libswresample-dev
  vlc-plugin-base vlc-plugin-visualization libavfilter7
Learn more about Ubuntu Pro at https://ubuntu.com/pro
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
debhelper is already the newest version (13.6ubuntu1).
iw is already the newest version (5.16-1build1).
libpcap-dev is already the newest version (1.10.1-4build1).
libsodium-dev is already the newest version (1.0.18-1build2).
python3-serial is already the newest version (3.5-1).
dh-python is already the newest version (5.20220403).
virtualenv is already the newest version (20.13.0+ds-2).
python3-all is already the newest version (3.10.6-1~22.04).
python3-future is already the newest version (0.18.2-5ubuntu0.1).
python3-pyroute2 is already the newest version (0.6.4-3ubuntu2).
python3-twisted is already the newest version (22.1.0-2ubuntu2.4).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.4).
The following packages were automatically installed and are no longer required:
  busybox-initramfs cpio initramfs-tools initramfs-tools-bin initramfs-tools-core klibc-utils libklibc linux-base linux-headers-5.15.0-102 linux-headers-5.15.0-102-generic usb-modeswitch
  usb-modeswitch-data zstd
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
rm -rf deb_dist
/home/daniel/wfb/wfb-ng/env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
make: /home/daniel/wfb/wfb-ng/env/bin/python: No such file or directory
make: *** [Makefile:50: deb] Error 127
```

lida2003, 2024-04-19:
Also tried:

```
sudo apt install python3-all libpcap-dev libsodium-dev python3-pip python3-pyroute2 \
            python3-future python3-twisted python3-serial python3-all-dev iw virtualenv \
            debhelper dh-python build-essential -y
sudo make deb
```

get error:

```
$ sudo make deb
g++  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
src/radiotap.c: In function ‘ieee80211_radiotap_iterator_init’:
src/radiotap.c:116:34: warning: taking address of packed member of ‘struct ieee80211_radiotap_header’ may result in an unaligned pointer va                                                      lue [-Waddress-of-packed-member]
  116 |         iterator->_next_bitmap = &radiotap_header->it_present;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
g++  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
cc  -Wall -O2 -DWFB_VERSION='"23.8.141.53708-b5db8d0e"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lpcap -lsodium
rm -rf deb_dist
/env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
make: /env/bin/python: No such file or directory
make: *** [Makefile:50: deb] Error 127
```

lida2003, 2024-04-19:
OK, fixed python3 path issue, and then I got:

```
rm -rf deb_dist
/home/daniel/wfb/wfb-ng/env/local/bin/python ./setup.py --command-packages=stdeb bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
make: *** [Makefile:50: deb] Error 1

```

lida2003, 2024-04-19:
> run `make env`

latest master run `make env`, which failed.

```
daniel@daniel-nvidia:~/wfb/wfb-ng$ make clean
rm -rf env wfb_rx wfb_tx wfb_keygen dist deb_dist build wfb_ng.egg-info wfb-ng-*.tar.gz _trial_temp *~ src/*.o
daniel@daniel-nvidia:~/wfb/wfb-ng$ make /home/daniel/wfb/wfb-ng/env
virtualenv --python=/usr/bin/python3 /home/daniel/wfb/wfb-ng/env
created virtual environment CPython3.10.12.final.0-64 in 309ms
  creator CPython3Posix(dest=/home/daniel/wfb/wfb-ng/env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/daniel/.local/share/virtualenv)
    added seed packages: pip==22.0.2, setuptools==59.6.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
/home/daniel/wfb/wfb-ng/env/bin/pip install --upgrade pip setuptools stdeb
make: /home/daniel/wfb/wfb-ng/env/bin/pip: No such file or directory
make: *** [Makefile:18: /home/daniel/wfb/wfb-ng/env] Error 127
daniel@daniel-nvidia:~/wfb/wfb-ng$ make deb
g++  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
src/radiotap.c: In function ‘ieee80211_radiotap_iterator_init’:
src/radiotap.c:116:34: warning: taking address of packed member of ‘struct ieee80211_radiotap_header’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  116 |         iterator->_next_bitmap = &radiotap_header->it_present;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lsodium -lpcap
g++  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lsodium
cc  -Wall -O2 -DWFB_VERSION='"24.4.8.62894-5b51172e"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lsodium
rm -rf deb_dist
/home/daniel/wfb/wfb-ng/env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
make: /home/daniel/wfb/wfb-ng/env/bin/python: No such file or directory
make: *** [Makefile:50: deb] Error 127
```

This is a python path fix based on stable branch (latest), which failed as what I have found "error: invalid command 'bdist_deb'"

```
daniel@daniel-nvidia:~/wfb/wfb-ng$ make clean
rm -rf env wfb_rx wfb_tx wfb_keygen dist deb_dist build wfb_ng.egg-info wfb-ng-*.tar.gz _trial_temp *~ src/*.o
daniel@daniel-nvidia:~/wfb/wfb-ng$ make /home/daniel/wfb/wfb-ng/env
virtualenv --python=/usr/bin/python3 /home/daniel/wfb/wfb-ng/env
created virtual environment CPython3.10.12.final.0-64 in 263ms
  creator CPython3Posix(dest=/home/daniel/wfb/wfb-ng/env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/daniel/.local/share/virtualenv)
    added seed packages: pip==22.0.2, setuptools==59.6.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
/home/daniel/wfb/wfb-ng/env/local/bin/pip install --upgrade pip setuptools stdeb
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /usr/local/lib/python3.10/dist-packages (24.0)
Requirement already satisfied: setuptools in /home/daniel/.local/lib/python3.10/site-packages (69.5.1)
Requirement already satisfied: stdeb in /home/daniel/.local/lib/python3.10/site-packages (0.10.0)
daniel@daniel-nvidia:~/wfb/wfb-ng$ make deb
g++  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu++11 -c -o src/rx.o src/rx.cpp
cc  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu99 -c -o src/radiotap.o src/radiotap.c
src/radiotap.c: In function ‘ieee80211_radiotap_iterator_init’:
src/radiotap.c:116:34: warning: taking address of packed member of ‘struct ieee80211_radiotap_header’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  116 |         iterator->_next_bitmap = &radiotap_header->it_present;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu99 -c -o src/fec.o src/fec.c
g++  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu++11 -c -o src/wifibroadcast.o src/wifibroadcast.cpp
g++ -o wfb_rx src/rx.o src/radiotap.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
g++  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu++11 -c -o src/tx.o src/tx.cpp
g++ -o wfb_tx src/tx.o src/fec.o src/wifibroadcast.o  -lrt -lpcap -lsodium
cc  -Wall -O2 -DWFB_VERSION='"23.8.263.77495-24722e9b"' -std=gnu99 -c -o src/keygen.o src/keygen.c
cc -o wfb_keygen src/keygen.o  -lrt -lpcap -lsodium
rm -rf deb_dist
/home/daniel/wfb/wfb-ng/env/local/bin/python ./setup.py --command-packages=stdeb bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
make: *** [Makefile:50: deb] Error 1
```

svpcom, 2024-04-19:
It seems that something broken in your python setup. Why it adds `local` to the path? Try to replace virtualenv to venv.
I.e instead of `virtualenv --python=/usr/bin/python3 /home/daniel/wfb/wfb-ng/env` run `/usr/bin/python3 -m venv /home/daniel/wfb/wfb-ng/env`

lida2003, 2024-04-19:
> It seems that something broken in your python setup. Why it adds `local` to the path? 

As `make deb` can't find the python, so I tried to fix it locally.

```
daniel@daniel-nvidia:~/wfb/wfb-ng$ git log -n 1
commit 24722e9b8b0dba0f202e064c3366715e3da5f052 (HEAD -> lida2003-test)
Author: Daniel Li <lida_mail@163.com>
Date:   Sat Apr 20 05:31:35 2024 +0800

    Fix for Jetson Orin Jimmy(22.04) python path
daniel@daniel-nvidia:~/wfb/wfb-ng$ git show 24722e9b8b0dba0f202e064c3366715e3da5f052
commit 24722e9b8b0dba0f202e064c3366715e3da5f052 (HEAD -> lida2003-test)
Author: Daniel Li <lida_mail@163.com>
Date:   Sat Apr 20 05:31:35 2024 +0800

    Fix for Jetson Orin Jimmy(22.04) python path

diff --git a/Makefile b/Makefile
index 82b9496..7c3e6ad 100644
--- a/Makefile
+++ b/Makefile
@@ -15,7 +15,7 @@ all: all_bin gs.key test

 $(ENV):
        virtualenv --python=$(PYTHON) $(ENV)
-       $(ENV)/bin/pip install --upgrade pip setuptools stdeb
+       $(ENV)/local/bin/pip install --upgrade pip setuptools stdeb

 all_bin: wfb_rx wfb_tx wfb_keygen

@@ -47,7 +47,7 @@ rpm:  all_bin $(ENV)

 deb:  all_bin $(ENV)
        rm -rf deb_dist
-       $(ENV)/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
+       $(ENV)/local/bin/python ./setup.py --command-packages=stdeb bdist_deb
        rm -rf wfb_ng.egg-info/ wfb-ng-$(VERSION).tar.gz

 bdist: all_bin
```

> Try to replace virtualenv to venv.
> I.e instead of `virtualenv --python=/usr/bin/python3 /home/daniel/wfb/wfb-ng/env` run `/usr/bin/python3 -m venv /home/daniel/wfb/wfb-ng/env`

I'm NOT sure what's going on here. I follow the guide `apt install python3.10-venv` and then `make clean` `make env` `make deb`
There is no luck on master or my test branch.

```
daniel@daniel-nvidia:~/wfb/wfb-ng$ /usr/bin/python3 -m venv /home/daniel/wfb/wfb-ng/env
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt install python3.10-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: /home/daniel/wfb/wfb-ng/env/bin/python3
```

svpcom, 2024-04-19:
Try without virtualenv/venv:
```
pip3 install --user stdeb
python3 ./setup.py --command-packages=stdeb bdist_deb
```

lida2003, 2024-04-19:
> Try without virtualenv/venv:
> 
> ```
> pip3 install --user stdeb
> python3 ./setup.py --command-packages=stdeb bdist_deb
> ```

OK, I have googled some issue, which might be related to specific python version. I'm NOT sure if it's python 3.10

```
daniel@daniel-nvidia:~/wfb/wfb-ng$ pip3 install --user stdeb
Requirement already satisfied: stdeb in /home/daniel/.local/lib/python3.10/site-packages (0.10.0)
daniel@daniel-nvidia:~/wfb/wfb-ng$ python3 ./setup.py --command-packages=stdeb bdist_deb
Traceback (most recent call last):
  File "/home/daniel/wfb/wfb-ng/./setup.py", line 39, in <module>
    setup(
  File "/home/daniel/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 104, in setup
    return distutils.core.setup(**attrs)
  File "/home/daniel/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 146, in setup
    _setup_distribution = dist = klass(attrs)
  File "/home/daniel/.local/lib/python3.10/site-packages/setuptools/dist.py", line 318, in __init__
    self.metadata.version = self._normalize_version(self.metadata.version)
  File "/home/daniel/.local/lib/python3.10/site-packages/setuptools/dist.py", line 354, in _normalize_version
    normalized = str(Version(version))
  File "/home/daniel/.local/lib/python3.10/site-packages/setuptools/_vendor/packaging/version.py", line 200, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
setuptools.extern.packaging.version.InvalidVersion: Invalid version: 'trunk'
```

svpcom, 2024-04-19:
To fix error with `trunk` just set correct VERSION and COMMON environment variables. See Makefile how to evaluate them:
```
COMMIT ?= $(shell git rev-parse HEAD)
VERSION ?= $(shell $(PYTHON) ./version.py $(shell git show -s --format="%ct" $(shell git rev-parse HEAD)) $(shell git rev-parse --abbrev-ref HEAD))
```

lida2003, 2024-04-19:
> To fix error with `trunk` just set correct VERSION and COMMIT environment variables.

Still not effective, please check commands as below:

```
$ cat test.sh
#!/bin/bash

VER_A=`git show -s --format="%ct"`
VER_B=`git rev-parse HEAD`
VER_C=`git rev-parse --abbrev-ref HEAD`

COMMIT=$VER_B
VERSION=`python3 ./version.py $VER_A $VER_B $VER_C`

echo "A:$VER_A B:$VER_B C:$VER_C "
echo "Version: $VERSION"
echo "Commit: $COMMIT"

daniel@daniel-nvidia:~/wfb/wfb-ng$ echo $VERSION
24.4.19.77495
daniel@daniel-nvidia:~/wfb/wfb-ng$ echo $COMMIT
24722e9b8b0dba0f202e064c3366715e3da5f052
daniel@daniel-nvidia:~/wfb/wfb-ng$ export VERSION=24.4.19.77495
daniel@daniel-nvidia:~/wfb/wfb-ng$ export COMMIT=24722e9b8b0dba0f202e064c3366715e3da5f052
daniel@daniel-nvidia:~/wfb/wfb-ng$ python3 ./setup.py --command-packages=stdeb bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
```

svpcom, 2024-04-19:
Run `sudo pip3 install stdeb` before this

lida2003, 2024-04-20:
> Run `sudo pip3 install stdeb` before this

daniel@daniel-nvidia:~/wfb/wfb-ng$ sudo pip3 install stdeb
[sudo] password for daniel:
Requirement already satisfied: stdeb in /usr/local/lib/python3.10/dist-packages (0.10.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
daniel@daniel-nvidia:~/wfb/wfb-ng$ python3 ./setup.py --command-packages=stdeb bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
daniel@daniel-nvidia:~/wfb/wfb-ng$ pip3 install stdeb
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: stdeb in /home/daniel/.local/lib/python3.10/site-packages (0.10.0)
daniel@daniel-nvidia:~/wfb/wfb-ng$ python3 ./setup.py --command-packages=stdeb bdist_deb
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'

lida2003, 2024-04-20:
@svpcom Thanks for the help!

Ok, I know the answer. My mistake:

```
-       $(ENV)/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
+       $(ENV)/local/bin/python ./setup.py --command-packages=stdeb bdist_deb
```

should be

```
$(ENV)/local/bin/python ./setup.py --command-packages=stdeb.command  bdist_deb
```

Rootcause is path issue. Maybe we have to come out a work around to fix jetson orin `make deb`.

closed 2024-04-20 by lida2003

ppydickbig, 2024-09-27:
run  /home/{usrname}/wfb/wfb-ng/env/bin/pip install stdeb    , it works