svpcom/wfb-ng #340: If RTL88xx support 11ac or only can use 11n to broadcast?

yuace, 2024-04-22:
hi,

I use wifibroadcast with RTL88xx, but I find only work in 11n.( Enable CONFIG_80211AC_VHT)
If RTL88xx support 11ac or only can use 11n to broadcast?

Thanks!

svpcom, 2024-04-22:
By default wfb_tx support only 11n injection. But you can patch it to allow 11ac (need to add additional flags to radiotap header if mcs_index is 11ac). Pull requests are welcome

See https://github.com/svpcom/rtl8812au/blob/60cb82df48247e94bae91c0642c17c90ad8d6276/core/rtw_xmit.c#L4335

yuace, 2024-04-22:
收到！！！

libc0607, 2024-05-08:
For anyone who wants to test VHT rates, I've made a dirty patch [here](https://github.com/libc0607/wfb-ng/commit/f7c8ea78dd8c8495bb22f184fcfd77152b25798a). Only tested on my 8812au & 8812eu.  

I'm not sure if I'm doing the mod in the right way ...

closed 2024-08-15 by svpcom

yuace, 2024-08-15:
收到！！！