svpcom/wfb-ng #341: [BUG]wifi driver failed to compile on ubuntu 24.04 LTS

mariarti, 2024-05-01:
**Describe the bug**
wifi driver failed to compile on ubuntu 24.04 LTS

```
/rtl8812au$ sudo ./dkms-install.sh 
About to run dkms install steps...
Deprecated feature: REMAKE_INITRD (/usr/src/rtl8812au-5.2.20.2/dkms.conf)
Creating symlink /var/lib/dkms/rtl8812au/5.2.20.2/source -> /usr/src/rtl8812au-5.2.20.2
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.2.20.2/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVER=6.8.0-31-generic KSRC=/lib/modules/6.8.0-31-generic/build..............(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.2.20.2 not found
Error! Bad return status for module build on kernel: 6.8.0-31-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/5.2.20.2/build/make.log for more information.
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.2.20.2/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVER=6.8.0-31-generic KSRC=/lib/modules/6.8.0-31-generic/build.............(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 5.2.20.2 not found
Error! Bad return status for module build on kernel: 6.8.0-31-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/5.2.20.2/build/make.log for more information.
Finished running dkms install steps.
Disabled IPv6 Successfuly 
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
```

**To Reproduce**
sudo ./dkms-install.sh

**Expected behavior**
success compile and finish

**Your setup (please complete the following information):**
 - OS Ubuntu 24.04 LTS amd64
 - Hardware i5 x86
 - WiFi card rtl8812au
 - WiFi driver rtl8812au-5.2.20.2
```
git clone -b v5.2.20 https://github.com/svpcom/rtl8812au.git
cd rtl8812au/
sudo ./dkms-install.sh
```

**Confirm you read**
- [x]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [x]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

svpcom, 2024-05-01:
Try to build without dkms (via `make` command) and upload build log

mariarti, 2024-05-01:
`Consult /var/lib/dkms/rtl8812au/5.2.20.2/build/make.log` for more information. from previous post. 
[dkms-install-log.txt](https://github.com/svpcom/wfb-ng/files/15181608/dkms-install-log.txt)

and build without dkms
[make.txt](https://github.com/svpcom/wfb-ng/files/15181616/make.txt)

svpcom, 2024-05-17:
Try https://github.com/svpcom/rtl8812au/commit/84cacef0af440de42839d7af31e4b6def96cb601 or latest master

mariarti, 2024-05-17:
[17-05-24.txt](https://github.com/svpcom/wfb-ng/files/15358432/17-05-24.txt)

svpcom, 2024-05-17:
Try build without dkms

mariarti, 2024-05-18:
[make-18-05-24.txt](https://github.com/svpcom/wfb-ng/files/15362539/make-18-05-24.txt)

svpcom, 2024-05-18:
Make succeeded. So you need to cleanup dkms sources (something like dkms uninstall or manually remove  /usr/src/rtl8812au-5.2.20.2)

mariarti, 2024-05-18:
[dkms.txt](https://github.com/svpcom/wfb-ng/files/15362729/dkms.txt)

svpcom, 2024-05-18:
Now dkms-install.sh should work

mariarti, 2024-05-18:
Yes, finally it performed without an errors.
```
user@pc:~/Downloads$ rm -rf rtl8812au/
user@pc:~/Downloads$ git clone -b v5.2.20 https://github.com/svpcom/rtl8812au.git
Cloning into 'rtl8812au'...
remote: Enumerating objects: 8659, done.
remote: Counting objects: 100% (192/192), done.
remote: Compressing objects: 100% (112/112), done.
remote: Total 8659 (delta 105), reused 143 (delta 79), pack-reused 8467
Receiving objects: 100% (8659/8659), 64.47 MiB | 21.51 MiB/s, done.
Resolving deltas: 100% (5899/5899), done.
user@pc:~/Downloads$ cd rtl8812au/
user@pc:~/Downloads/rtl8812au$ sudo ./dkms-install.sh 
[sudo] password for user: 
About to run dkms install steps...
Error! DKMS tree already contains: rtl8812au-5.2.20.2
You cannot add the same module/version combo more than once.
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.2.20.2/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVER=6.8.0-31-generic KSRC=/lib/modules/6.8.0-31-generic/build..................................
Signing module /var/lib/dkms/rtl8812au/5.2.20.2/build/88XXau_wfb.ko
Cleaning build area...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.2.20.2/source/dkms.conf)

88XXau_wfb.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-31-generic/updates/dkms/
depmod....
Finished running dkms install steps.
Disabled IPv6 Successfuly 
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1

user@pc:~$ sudo modprobe 88XXau_wfb
[sudo] password for user: 
user@pc:~$ lsmod
Module                  Size  Used by
88XXau_wfb           2404352  0
```
Later will go further with receiving data.
Thanks a lot for fix and help!

closed 2024-05-18 by mariarti