svpcom/wfb-ng #349: [BUG] Unable to install wfb-ng in fresh ubuntu installation

ramkrishna757575, 2024-06-15:
**Describe the bug**
Installation of wfb-ng in a fresh instance of ubuntu fails when following the installation instructions:
[image]

When running the last command of `sudo ./scripts/install_gs.sh {My WLAN Interface}`, the following error seems to occur:

```
/home/ubuntu/wfb-ng/env/bin/python ./setup.py --command-packages=stdeb.command bdist_deb
running bdist_deb
running sdist_dsc
running egg_info
creating wfb_ng.egg-info
writing wfb_ng.egg-info/PKG-INFO
writing dependency_links to wfb_ng.egg-info/dependency_links.txt
writing entry points to wfb_ng.egg-info/entry_points.txt
writing top-level names to wfb_ng.egg-info/top_level.txt
writing manifest file 'wfb_ng.egg-info/SOURCES.txt'
reading manifest file 'wfb_ng.egg-info/SOURCES.txt'
adding license file 'LICENSE.txt'
writing manifest file 'wfb_ng.egg-info/SOURCES.txt'
Traceback (most recent call last):
  File "/home/ubuntu/wfb-ng/./setup.py", line 39, in <module>
    setup(
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/__init__.py", line 103, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 184, in setup
    return run_commands(dist)
           ^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
    dist.run_commands()
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/dist.py", line 968, in run_command
    super().run_command(command)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/command/bdist_deb.py", line 34, in run
    self.run_command('sdist_dsc')
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
    self.distribution.run_command(command)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/dist.py", line 968, in run_command
    super().run_command(command)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/command/sdist_dsc.py", line 30, in run
    debinfo = self.get_debinfo()
              ^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/command/common.py", line 197, in get_debinfo
    debinfo = DebianInfo(
              ^^^^^^^^^^^
  File "/home/ubuntu/wfb-ng/./setup.py", line 13, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/util.py", line 802, in __init__
    check_cfg_files(cfg_files, module_name)
  File "/home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/util.py", line 733, in check_cfg_files
    cfg = ConfigParser.SafeConfigParser()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
make: *** [Makefile:50: deb] Error 1
```

**To Reproduce**
1. Install a fresh copy of ubuntu 22 or 24
2. Follow the wfb-ng installation guide here https://github.com/svpcom/wfb-ng

**Expected behavior**
Installation of wfb-ng should succeed without any errors.

**Your setup (please complete the following information):**
 - OS : Ubuntu 24.04 LTS
 - Hardware : x86
 - WiFi cards : RTL8812AU
 - WiFi drivers : patched driver from https://github.com/svpcom/rtl8812au.git

**Confirm you read**
- [ Yes]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [ Yes]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

svpcom, 2024-06-16:
What python version do you have?

ramkrishna757575, 2024-06-16:
> What python version do you have?

My system has Python 3.12.3

doveshots, 2024-06-17:
Just vim /home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/util.py
rewrite SafeConfigPaser in line 733 to ConfigParser could work for me
also rewrite all readfp() to read_file(),and change the filename=filename to only filename

ramkrishna757575, 2024-06-17:
> Just vim /home/ubuntu/wfb-ng/env/lib/python3.12/site-packages/stdeb/util.py rewrite SafeConfigPaser in line 733 to ConfigParser could work for me also rewrite all readfp() to read_file(),and change the filename=filename to only filename

Thanks @doveshots .
I followed your instructions and was able to successfully build it. Seems like the installation went through successfully.
However the wifibroadcast service seems to be crashing.

```
> systemctl status wifibroadcast@gs
● wifibroadcast@gs.service - WFB profile gs
     Loaded: loaded (/usr/lib/systemd/system/wifibroadcast@.service; disabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2024-06-17 11:46:11 IST; 4s ago
    Process: 17290 ExecStart=/usr/bin/wfb-server gs ${WFB_NICS} (code=exited, status=1/FAILURE)
   Main PID: 17290 (code=exited, status=1/FAILURE)
        CPU: 267ms
```

doveshots, 2024-06-19:
I read the document of this lib,it seems like it's using python2.7,what about using python-is-python2 or reinstalling older os,my pi is not around,sorry for not being able to test.it seems like the project wiki should be updated or something,even the name of the folder is wrong :d

svpcom, 2024-06-19:
Try to install stdeb from this repo https://github.com/DdangJin/stdeb

ramkrishna757575, 2024-06-20:
Thanks @svpcom . Seems like the installation worked fine. But the wifibroadcast service is crashing for some reason. Is there a way to check the error logs?

[image]

svpcom, 2024-06-20:
`sudo journalctl -xu wifibroadcast@gs`
see troubleshooting guide in wiki

ramkrishna757575, 2024-06-22:
> readfp

This is what I'm getting when I run  `sudo journalctl -xu wifibroadcast@gs`:
```
Jun 17 11:40:51 X570-PC wfb-server[11675]: Traceback (most recent call last):
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/config_parser.py", line 77, in parse_>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     config.readfp(fd, filename=filename)
Jun 17 11:40:51 X570-PC wfb-server[11675]:     ^^^^^^^^^^^^^
Jun 17 11:40:51 X570-PC wfb-server[11675]: AttributeError: 'RawConfigParser' object has no attribute 'readfp'. Did you mean: '>
Jun 17 11:40:51 X570-PC wfb-server[11675]: During handling of the above exception, another exception occurred:
Jun 17 11:40:51 X570-PC wfb-server[11675]: Traceback (most recent call last):
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/bin/wfb-server", line 33, in <module>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     sys.exit(load_entry_point('wfb-ng==23.8.141.53708', 'console_scripts', 'wfb-ser>
Jun 17 11:40:51 X570-PC wfb-server[11675]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/bin/wfb-server", line 25, in importlib_load_entry_point
Jun 17 11:40:51 X570-PC wfb-server[11675]:     return next(matches).load()
Jun 17 11:40:51 X570-PC wfb-server[11675]:            ^^^^^^^^^^^^^^^^^^^^
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
Jun 17 11:40:51 X570-PC wfb-server[11675]:     module = import_module(match.group('module'))
Jun 17 11:40:51 X570-PC wfb-server[11675]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
Jun 17 11:40:51 X570-PC wfb-server[11675]:     return _bootstrap._gcd_import(name[level:], package, level)
Jun 17 11:40:51 X570-PC wfb-server[11675]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap_external>", line 995, in exec_module
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/server.py", line 36, in <module>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     from .common import abort_on_crash, exit_status, df_sleep
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/common.py", line 24, in <module>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     from .conf import settings
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/conf/__init__.py", line 17, in <modul>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     settings, cfg_files = _parse_config()
Jun 17 11:40:51 X570-PC wfb-server[11675]:                           ^^^^^^^^^^^^^^^
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/conf/__init__.py", line 14, in _parse>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     return config_parser.parse_config(os.path.join(os.getcwd(), os.path.dirname(__f>
Jun 17 11:40:51 X570-PC wfb-server[11675]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^>
Jun 17 11:40:51 X570-PC wfb-server[11675]:   File "/usr/lib/python3/dist-packages/wfb_ng/config_parser.py", line 79, in parse_>
Jun 17 11:40:51 X570-PC wfb-server[11675]:     raise ConfigError(v)
Jun 17 11:40:51 X570-PC wfb-server[11675]: wfb_ng.config_parser.ConfigError: 'RawConfigParser' object has no attribute 'readfp'
Jun 17 11:40:51 X570-PC systemd[1]: wifibroadcast@gs.service: Main process exited, code=exited, status=1/FAILURE
```

I am not familiar with Python. Not sure why the build contains the incorrect function name.

ramkrishna757575, 2024-06-24:
ok..i was able to fix the above error. Install is working.

Now the service is crashing. When i run `sudo service wifibroadcast@gs status`

This is the error I see:

```
○ wifibroadcast@gs.service - WFB profile gs
     Loaded: loaded (/usr/lib/systemd/system/wifibroadcast@.service; disabled; preset: enabled)
     Active: inactive (dead)

Jun 24 16:51:34 X570-PC wfb-server[9506]: [defer.Deferred._runCallbacks #info] # nmcli device show wlx502b73d146e8
Jun 24 16:51:34 X570-PC wfb-server[9506]: [defer.Deferred._runCallbacks #info] # ip link set wlx502b73d146e8 down
Jun 24 16:51:35 X570-PC wfb-server[9506]: [defer.Deferred._runCallbacks #info] # iw dev wlx502b73d146e8 set monitor otherbss
Jun 24 16:51:35 X570-PC wfb-server[9506]: [server.init_wlans #error] RTNETLINK answers: Operation not possible due to RF-kill
Jun 24 16:51:35 X570-PC wfb-server[9506]: [common.abort_on_crash #info] Stopping reactor due to fatal error: RC 2: ip link set wlx502b73d146e8 up
Jun 24 16:51:35 X570-PC wfb-server[9506]: [base.EPollReactor.mainLoop #info] Main loop terminated.
Jun 24 16:51:35 X570-PC wfb-server[9506]: [server.main #info] Exiting with code 1
Jun 24 16:51:35 X570-PC systemd[1]: wifibroadcast@gs.service: Main process exited, code=exited, status=1/FAILURE
Jun 24 16:51:35 X570-PC systemd[1]: wifibroadcast@gs.service: Failed with result 'exit-code'.
Jun 24 16:51:39 X570-PC systemd[1]: Stopped wifibroadcast@gs.service - WFB profile gs.
```

svpcom, 2024-06-24:
See Setup HOWTO how to unblock rfkill

ramkrishna757575, 2024-06-24:
Thanks @svpcom . I was able to resolve the issue. Seems like the rtl8812u driver was not working correctly.
I reinstalled it and it is working now.

closed 2024-06-24 by svpcom