svpcom/wfb-ng #350: The fopen function can specify binary read mode, which is more suitable for reading key files.

TalusL, 2024-06-18:
https://github.com/svpcom/wfb-ng/blob/7180ffb3f74f095dcede5b389ce975909d87cace/src/rx.cpp#L269

The fopen function can specify binary read mode, which is more suitable for reading key files. 

```c
fopen(keypair.c_str(), "rb") 
```

svpcom, 2024-06-18:
It is only for non-posix systems:

> The mode string can also include the letter 'b' either as a last character or as a character between the characters in any of the two-character strings described above.  This is strictly for compatibility with ISO C and has no effect; the 'b' is ignored on all POSIX conforming systems, including Linux.  (Other systems may treat text files and binary files differently, and adding the  'b'  may be a good idea if you do I/O to a binary file and expect that your program may be ported to non-UNIX environments.)

TalusL, 2024-06-18:
> It is only for non-posix systems:

YES，I encountered a bug when porting this code to Windows. Thank for the information.

closed 2024-06-18 by TalusL