svpcom/wfb-ng #361: ubnutu重启后无法正常工作

caaagar, 2024-08-10:
Ubnutu正常安装8812EU驱动后识别出了8812EU并可以连接WIFI，在安装wfb(删除了key生成与复制，并从摄像头端复制GS.KEY到ETC目录下)后可以正常工作
[image]
在重启后使用WFB-CLI GS就无法正常工作

caaagar, 2024-08-10:
[image]

caaagar, 2024-08-10:
[image]

caaagar, 2024-08-10:
[image]

svpcom, 2024-08-10:
`sudo journalctl -xu wifibroadcast@gs > debug.log` and upload it here

caaagar, 2024-08-10:
顺便一说Ubnutu安装时需要修改某个文件的3处代码

caaagar, 2024-08-10:
> `sudo journalctl -xu wifibroadcast@gs > debug.log`并上传至此处

过几天看一下，明天休息设备在公司

caaagar, 2024-08-10:
> `sudo journalctl -xu wifibroadcast@gs > debug.log`并上传至此处

明天休息设备在公司