svpcom/wfb-ng #363: 服务重启后无法正常启动

caaagar, 2024-08-13:
[debug.log](https://github.com/user-attachments/files/16593539/debug.log)

svpcom, 2024-08-13:
It seems that there are some processes (wpa_supplicant, dhclient, etc) that touch wlx387acc6b1033 interface
```
8月 09 14:58:14 youg-N9x0TD-TF wfb-server[6579]: [server.DbgProtocol.lineReceived #info] mavlink rx: Error: socket error!
8月 09 14:58:14 youg-N9x0TD-TF wfb-server[6579]: [server.DbgProtocol.lineReceived #info] tunnel rx: Error: socket error!
```

See Setup HOWTO in wiki how to disable them

P.S. Please do comments in English, also add some description of your setup

caaagar, 2024-08-13:
I referred to the content from 'Disable wpa_supplicant and other daemons on WFB-NG wlan interface!' for the configuration, so all the settings are in place and correct.

svpcom, 2024-08-13:
Check dmesg output and ensure that there are no disconnects wifi cards from usb

caaagar, 2024-08-13:
> 检查 dmesg 输出，确保 wifi 卡没有从 usb 断开连接

问题以解决，我是傻子
重启后要用systemctl start wifibroadcast@gs来启动服务

svpcom, 2024-08-13:
Use `systemctl enable wifibroadcast@gs` and `systemctl enable wifibroadcast` to start it automatically after boot

closed 2024-08-15 by svpcom