svpcom/wfb-ng #373: [BUG] install_gs.sh failing: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'

StephenCarlson, 2024-09-27:
**Describe the bug**
I'm following the instruction for the Ubuntu Ground Station, everything is working up to `sudo ./scripts/install_gs.sh <my_specific_wifi_alias_here>`. This is the error I'm getting:

```
virtualenv --python=/usr/bin/python3 /home/steve/dev/wfb-ng/env
Traceback (most recent call last):
  File "/usr/bin/virtualenv", line 5, in <module>
    from virtualenv.__main__ import run_with_catch
  File "/usr/lib/python3/dist-packages/virtualenv/__init__.py", line 3, in <module>
    from .run import cli_run, session_via_cli
  File "/usr/lib/python3/dist-packages/virtualenv/run/__init__.py", line 14, in <module>
    from .plugin.creators import CreatorSelector
  File "/usr/lib/python3/dist-packages/virtualenv/run/plugin/creators.py", line 6, in <module>
    from virtualenv.create.via_global_ref.builtin.builtin_way import VirtualenvBuiltin
  File "/usr/lib/python3/dist-packages/virtualenv/create/via_global_ref/builtin/builtin_way.py", line 7, in <module>
    from virtualenv.create.creator import Creator
  File "/usr/lib/python3/dist-packages/virtualenv/create/creator.py", line 15, in <module>
    from virtualenv.discovery.cached_py_info import LogCmd
  File "/usr/lib/python3/dist-packages/virtualenv/discovery/cached_py_info.py", line 23, in <module>
    _CACHE[Path(sys.executable)] = PythonInfo()
  File "/usr/lib/python3/dist-packages/virtualenv/discovery/py_info.py", line 86, in __init__
    self.distutils_install = {u(k): u(v) for k, v in self._distutils_install().items()}
  File "/usr/lib/python3/dist-packages/virtualenv/discovery/py_info.py", line 161, in _distutils_install
    i.finalize_options()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/command/install.py", line 67, in finalize_options
    super().finalize_options()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/command/install.py", line 408, in finalize_options
    'dist_fullname': self.distribution.get_fullname(),
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 267, in get_fullname
    return _distribution_fullname(self.get_name(), self.get_version())
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 285, in _distribution_fullname
    canonicalize_version(version, strip_trailing_zero=False),
TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
```

**To Reproduce**
- From an Ubuntu 22.04 LTS install, follow the normal installation procedure per the README.md: https://github.com/svpcom/wfb-ng?tab=readme-ov-file#quick-start-using-debian-or-ubuntu-ground-station
- I've tried various means to fix the issue, such as `python -m pip install --upgrade pip setuptools`, or downgrading the setuptools package via `python -m pip install setuptools==70.3.0`. No other steps attempted except `sudo make clean` between attempts at executing `sudo ./scripts/install_gs.sh wlx00c0cab3e254`
- I have also tried different versions of the project, with `git checkout master`, `git checkout stable`, or other specific releases, with appropriate `git reset --hard HEAD` or equivalent commands to clean the repo state.

**Expected behavior**
Normal installation per the README.md instructions.

**Your setup (please complete the following information):**
 - Ubuntu 22.04 LTS
 - x86
 - WiFi card not relevant in this case, but 8812au if this must be relevant
 - WiFi drivers from https://github.com/svpcom/rtl8812au.git

Additional configuration notes:
Python 3.10.12
pip 24.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Python setuptools version 70.3.0, downgraded from 75.1.0 (I've tried both, downgrading does nothing to correct the issue)

**Additional context**
I've found the following hints as to why things are breaking, it seems they have changed the behavior by design:
https://github.com/pypa/setuptools/issues/4483
https://github.com/pypa/setuptools/issues/4496

Also, I've found this closed issue from a few weeks ago, for a similar issue with the build system:
https://github.com/svpcom/wfb-ng/issues/349

**Confirm you read**
- [X]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [X]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

svpcom, 2024-09-27:
What version of wfb-ng do you use? Becase latest wfb-ng doesn't have `virtualenv --python=...` in the Makefile. It now have `python3 -m virtualenv ...`

svpcom, 2024-09-27:
And it build successfully on ubuntu-22.04: https://github.com/svpcom/wfb-ng/actions/runs/11059494849

StephenCarlson, 2024-09-27:
@svpcom I tried compiling on `stable`, `master` and the current release. Here is what I ended up doing: I went to the source file that was causing the issue, `/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py`, and I commented out the offending line and saved with `canonicalize_version(version),` . This is absolutely terrible and is not the correct way to solve the issue on my side, but at least I got it to compile the package and it continued to finish, and I've been playing with the working system since. I hope this isn't just my Ubuntu 22 environment; has anyone else run into this? On a side note, I just about trashed my Ubuntu install by installing Python 3.12 and then trying the update-alternatives mechanism to modify the default python instance, to great disaster, but I was able to revert back.

StephenCarlson, 2024-09-29:
@svpcom Indeed, but please be advised, it was not working on 3.10, prior to my tampering with installing other Python versions. The Ubuntu 22.04 install was fairly fresh; I don't think I had broken the Python libraries prior. I'll report here if I can reproduce the issue on a clean OS install or separate machine, hopefully if this is a reproducible bug, others will find this and report, perhaps I am just the first to report here?

svpcom, 2024-09-30:
@StephenCarlson wfb-ng has support for deb build using docker. For example you can run: `make deb_docker DOCKER_SRC_IMAGE=ubuntu:22.04` and ensure that in clean environment (docker image for ubuntu 22.04) it will build without problems