svpcom/wfb-ng #378: [BUG] 8812eu not changing wifi channels

aepkolol, 2024-10-22:
**Describe the bug**
I have a 8812au on the gs side which works perfectly, however the 8812eu on the drone refuses to change wifi channels. I can confirm the wifibroadcast@drone service is running and sent the iw command to change to channel 161. However it is stuck locked to channel 1.  However it does change to monitor mode. Trying to manually change the channel to 161 gives me this error:

```iw dev wlan0 set channel 161 HT20command failed: Operation not permitted (-1)```

In the wiki it notes 

```Check with ethtool -i wlanXX that drivers version is empty (it will equal to kernel version for stock driver and empty for patched driver). NVIDIA Jetson has stock rtl8812au installed. You need to remove it!```

however the driver version is not empty, like it is on the gs side with the 8812au. It's reporting the driver as 

```88x2eu```

where the main readme says it should be "8812eu" I followed the instructions of this repo exactly, except for this part from the drivers git repo to get it to compile

```
Or, for arm64, run:

sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
```

I forget the iw command I used, but it listed all the available wifi channels and most outside of a small handful in the 2.4ghz range were listed as "disabled" however it correctly lists my region as BO, and its set as such in the .cfg. 

Now, the weird thing is.. I had this working a few days before. I just reflashed the jetson and followed the wiki to the letter. I'm not sure what has changed since, and I can't get it working to where it was before. So I am hoping someone can shed some light on where to go from here, otherwise I'd assume it was a driver/kernel issue. 

**To Reproduce**
Install as per the readme.md and wiki

**Expected behavior**
Connection to gs

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Your setup (please complete the following information):**
 - OS [e.g. Ubuntu 18.04, Raspbian]
 ```L4T 35.6.0 - Ubuntu 20.04 - 5.10.216-tegra```
 - Hardware [e.g. x86, rpi, nanopi neo2 ]
 ```Nvidia Jetson Orin Nano```
 - WiFi cards [e.g. rt28xx, rtl8812au]
 ```rtl8812eu```
 - WiFi drivers [e.g. stock, patched from http://.. ]
https://github.com/svpcom/rtl8812eu/
**Additional context**
Add any other context about the problem here.

**Confirm you read**
- [ X]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [ X]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

svpcom, 2024-10-22:
It seems this channel is blocked by your CRDA rules. Run:
1. `iw reg get`
2. `iw phy`

aepkolol, 2024-10-22:
```
spiri@mu2:~$ iw reg get
global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR

spiri@mu2:~$ sudo ethtool -i wlan0
[sudo] password for spiri:
driver: rtl88x2eu
version: v5.15.0.1-197-ge90764eb9.202311
firmware-version: 1.15
expansion-rom-version:
bus-info: 1-3.3:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
spiri@mu2:~$ sudo iw dev
phy#1
        Interface wlan1
                ifindex 8
                wdev 0x100000001
                addr d8:44:89:5c:09:8a
                ssid BELL676
                type managed
                txpower 12.00 dBm
phy#0
        Interface wlan0
                ifindex 7
                wdev 0x1
                addr 78:22:88:19:2a:63
                type managed
                txpower 42949572.96 dBm
spiri@mu2:~$ sudo iw dev wlan1 set channel 161 HT20
command failed: Device or resource busy (-16)
```

wlan1 is an 8812bu I have attached to ssh into the drone over wifi

svpcom, 2024-10-22:
@aepkolol 
```
phy#0
        Interface wlan0
                ifindex 7
                wdev 0x1
                addr 78:22:88:19:2a:63
                type managed
```
it should be in monitor mode

svpcom, 2024-10-22:
`Device or resource busy (-16)` check that no other processes (wpa_supplicant, dhclient, etc) use this card.

aepkolol, 2024-10-22:
Sorry, I realize I did 

```sudo iw dev wlan1 set channel 161 HT20```

but that was targeting wlan1, which I use for normal wifi. Doing it for wlan0 outputs nothing to the terminal, but the channel stays the same. 

```
spiri@mu2:~$ sudo iw dev wlan0 set channel 161 HT20
spiri@mu2:~$ iw wlan0 info
Interface wlan0
        ifindex 7
        wdev 0x1
        addr 78:22:88:19:2a:63
        type monitor
        wiphy 0
        channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
        txpower 18.00 dBm
        
```

svpcom, 2024-10-22:
this is normal. It switch channel, but driver doesn't report it (driver bug, you can fill ticket to driver author)

aepkolol, 2024-10-25:
If I connect to a normal network, I can see with ```iw dev``` that it changes channels properly. It just seems like setting the channel manually with ```sudo iwconfig wlan0 channel 161``` refuses to work. Even if the driver isn't reporting the channel change, I'm unable to see a connection on either side of wfb. No log message mentioning bad encryption keys, etc. It's just as if it isn't pushing a signal out. 

I can see in wfb-cli drone and gs that packets are being injected on the tx side however. Just neither are receiving.