svpcom/wfb-ng #379: wfb-ng Download configuration problems to Ubuntu22.04

dangbo08, 2024-10-25:
May I ask, I configured wfb-ng on Ubuntu22.04, and downloaded the driver of the network adapter, I want to use the Ubuntu virtual machine as a ground station to receive openipc camera data, although my network adapter received pkt, but I checked, there is no data in the pkt, what is going on? Can someone tell me, thank you very much!!
[image]

dangbo08, 2024-10-25:
I generated dron.key and gs.key in the openipc camera, then gave the gs.key to the Ubuntu ground station, and configured the WiFi channel to be the same, but it still didn't work.

svpcom, 2024-10-25:
Try to update GS to the latest version. It may be version incompatibility between drone and gs

dangbo08, 2024-10-28:
[image]
Hello, I downloaded the new wfb-ng, but the result is still the same, I would like to ask whether wfb-ng must correspond to its version? Or is it backward compatible? Because I downloaded the ground station firmware from openipc's official website, burning it into nvr can connect with my openipc camera and display images.
[image]
[image]

dangbo08, 2024-10-28:
[image]
Hello, I have been able to display pictures, but the delay and packet loss problems are very serious, is this situation related to my WiFi network card?

svpcom, 2024-10-28:
yes. A lot of lost packets shows that there is a problem with wifi card and/or usb bus. Do you use native linux or run it in the virtual machine?

dangbo08, 2024-10-29:
I ran this on a virtual machine.

svpcom, 2024-10-29:
This is a known problem described in SetupHOWTO. Use native linux