svpcom/wfb-ng #380: [REQUEST] Dynamic switching hooks for link quality

lida2003, 2024-10-31:
I didn't find dynamic switching features for improving the link quality (packet loss).

This kind of dynamic tuning features should be useful for applications, such as FPV video feed. 

e.g. When flying an FPV

It will fly low altitude or fly behind obstacle(trees, walls etc) for a short period of time. Then the application need to:

- increase the transmitting power
- enlarge the bandwidth
- decrease the bitrate
- etc

in order to get a stable feed without any glitches if possible.

And wfb-ng is a communication foundation of the FPV video trasmittion backbone tunnel. Hope there are hooks or interfaces for dynamic tunnings.

Then there are chances for link quality management. what do you think?

PS: I might be wrong, please correct me if I missed something.

svpcom, 2024-10-31:
They already exists. See `wfb_tx_cmd`. Also you can connect to telemetry socket (like wfb-cli does) and get full link state (include rssi for every antenna, lost and recovered packets, etc). This API already used by openipc adaptive link https://www.youtube.com/watch?v=9WXtJEcyyeI

svpcom, 2024-10-31:
https://github.com/sickgreg/OpenIPC-Adaptive-Link

closed 2024-10-31 by lida2003