svpcom/wfb-ng #381: `RTL8812BU` Launch Report

MuratovAS, 2024-11-10:
Initial conditions:
- transceiver: tp-link archer t4u plus (`RTL8812BU`)
- wfb-ng: v24.08.2
- linux: kali-2024.3-qemu-amd64
- driver: see below
- tools: iPerf3

Result:
After a series of trials and errors I managed to achieve operability on the `RTL8812BU` chip. The system works in two ranges, `2.4Ghz` and `5Ghz`.

I checked the functionality with the `rtw88_8822bu` driver built into the kernel. With default settings the system refused to work, data is not sent or received.
After I changed the line to `frame_type = 'rts'`, the connection worked, but the transfer speed was unacceptably low, about `50Kb/s`.

I was able to achieve stable operation using a [custom kernel module from RinCat](https://github.com/RinCat/RTL88x2BU-Linux-Driver). Using this code, the connection speed through the tunnel was about `8Mb/s`.

Based on the data I received, the chip can be used. As soon as I have more data, I will let you know here.

svpcom, 2024-11-10:
@MuratovAS Thanks!

P.S. I've seen your portfolio - could you advice dc-dc for powering such wifi cards? Most of powerful 8812au and 8812eu cards consume > 3A in pulse when transmitting and generic dc-dc converters from aliexpress requires additional low-esr capacitors to survive these pulses. Are any better (an cheap) solution to power such cards from 3S-6S batteries?

MuratovAS, 2024-11-11:
I would advise:
- Take large cross-section wires, multi-wire (многопроволочный) (reduce AC/DC resistance)
- Power wires should be well twisted together.
- Make this wire as short as possible, a couple of centimeters.

I would build a converter on `RT6365`, simple, reliable, cheap, powerful.
If you need a ready-made solution `URB2405LD-20WR3` or `URB2405LD-30WR3`

svpcom, 2024-11-11:
@MuratovAS Thanks!
I had very bad experience with this dc-dc:
[image]

It dies with smoke when connected to 24V and connection is unreliable:  https://t.me/wfb_ng/4457/11336

svpcom, 2024-11-12:
@MuratovAS Do you have any idea why glitching of input voltage can lead to dc-dc burnout? It is IC design flaw or bad schematics? All three dc-dc in my test dies identically:
[image]
case near +Vin pin was burn out

voanhcuoc, 2024-11-13:
Thanks to this report, I've conducted an experiment:
* transceiver: ASUS USB-AC53 Nano
* wfb-ng: v24.08 (official pi 3 image)
* driver: RinCat
* linux: Raspberry Pi 3
* camera: Chinese no-name webcam (640x480)
* stbc = 2
* tx power : default

I don't have DSI monitor for Pi, so I connected an Android phone with the Pi by a USB cable. Then, I used `adb reverse` to tunnel RTSP connection over USB in order to show the stream in Android VLC.

The test is conducted on the outdoor ground with no obstacle.

Result:
The video is fairly good at close range. It start to pixelate at 15m and then lose connection at 25m.  
Pretty bad result, I guess due to limit of small onboard antennas, AC53 Nano is a smallest-possible USB wifi, anyway.

MuratovAS, 2024-11-14:
> @MuratovAS Do you have any idea why glitching of input voltage can lead to dc-dc burnout? It is IC design flaw or bad schematics? All three dc-dc in my test dies identically:

I am seeing the power mosfet burn out. IC: `tps5430`
Possible causes:
- Transistor not fully opened (transistor control system did not have time to enter operating mode)
- Load starting current is excessively high (output capacitance is too high + load is too powerful).
- Internal oscillatory processes have occurred in the system.

Solution:
- Do not load DCDC at full power at the moment of starting (in reality, this is a very rare scenario)!
- Choose an IC for DCDC with a soft-start function

P.S. I have never used this power supply at a current greater than 2A. The recommended current is 1.5A

svpcom, 2024-11-14:
@MuratovAS The main problem is that most of wfb-ng users are hobby users and unable to make custom PCB design themself. Most of ready to buy dc-dc from aliexpress has awful design and overstated parameters. It would be great if someone who specializes in this topic would make such an opensource design, which could be easily repeated by yourself or ordered on pcbway