svpcom/wfb-ng #387: How to cross-compile?

dangbo08, 2024-11-27:
Hello, I want to cross-compile this project into my arm single chip computer to achieve the effect of nvr in openipc. Since the video display interface of nvr is not the effect I want, I want to cross-compile and then redesign the video display and reasonably plan the data display of flight control. However, I always encounter the problem of cross-compilation environment when compiling, so I have been unable to succeed, so I would like to ask if you have relevant solutions, please teach me if you have, thank you very much ！！！

dangbo08, 2024-12-09:
I cross-compile according to your method, but there are some errors in the end, so I asked chatgpt about the reason for the port. Then I checked that the generated binary file is indeed a file that can run on the arm platform. May I ask how I should put it on the embedded device? On Linux x86 systems there is a systemctl start wifibroadcast@gs command to boot, how to boot on embedded platforms?
[image]
[image]
Above is the compilation of the final error picture, and the format of the binary file. Is my operation correct? If there are any mistakes, please comment, thank you very much ！！！！

svpcom, 2024-12-09:
Use bunded qemu or disable tests during build, because standard qemu doen't implement some system calls needed for test run. 
```
Docker cross build requires patched QEMU!
Apply ./scripts/qemu/qemu.patch to qemu-7.2.0 and build it:
  ./configure --prefix=/opt/qemu --static --disable-system && make && sudo make install
```

dangbo08, 2024-12-11:
Hello, may I ask, if my arm system supports the WiFi network card and the related dependency library of wfb, do I only need to cross-compile the three files wfb_tx.wfb_rx,wfb_keygen, and then I can design the relevant ways of receiving and sending files? Thank you!!

svpcom, 2024-12-11:
wfb-ng deals with udp packets. If you want to send files then you have two ways:
1. Setup ipv4 tunnel on top of wfb-ng. If you use full version (with python) it works  out of box. If you want to use C-only then you need to setup it yourself using wfb_tun program. See OpenIPC projects sources for reference
2. Write your own program and implement custom filter transfer protocol on top of udp

> Hello, may I ask, if my arm system supports the WiFi network card and the related dependency library of wfb, do I only need to cross-compile the three files wfb_tx.wfb_rx,wfb_keygen, and then I can design the relevant ways of receiving and sending files? Thank you!!

dangbo08, 2025-01-03:
Hello, I cross-compiled your project, put the generated binary files wfb_keygen,wfb_tx,wfb_rx,wfb_tun,wfb_tx_cmd these five binary files into my environment to use, and copied the openipc edit related script files such as wifibroadcast, I ran the file and did not receive data and was prompted that I had error data. May I ask if the five binaries migrated above can only use them to receive data? Thank you!!
[image]
2 block overrides
3 packets lost
41 block overrides
15603 packets lost
40 block overrides
15335 packets lost
40 block overrides
14435 packets lost
[image]

[image]

svpcom, 2025-01-03:
Use full version (deb packet from `make deb`). I don't know how openipc startup scripts work

dangbo08, 2025-01-08:
I recompiled my arm environment, downloaded some related configuration files, and then compiled your project in this system has been successful, thank you very much for the ideas provided before, and the understanding of your project has deepened a lot, thank you very much!!
[image]