svpcom/wfb-ng #397: [BUG] wfb-ng-cluster-manager ssh uses bash

aepkolol, 2025-01-12:
**Describe the bug**
When running the wfb-cluster-manager service with the --ssh flag, it successfully connects to the openWRT router over ssh, however it tries to use bash [here](https://github.com/svpcom/wfb-ng/blob/4610a7110aa0326956a65e43e105092e06cfc631/wfb_ng/server.py#L152C1-L160C59) causing it to fail.

```
                if ssh_user and ssh_port:
                    dl.append(SSHClientProtocol(node,
                                                ssh_user,
                                                '/bin/bash',
                                                key=ssh_key,
                                                port=ssh_port,
                                                use_agent=ssh_key is None,
                                                stdin=setup_script).start()\
                              .addBoth(_ssh_exited, node))

```
when busybox's shell uses ash.

**To Reproduce**
```sudo systemctl start wfb-cluster-manager@```

**Expected behavior**
Connect to the shell successfully 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Your setup (please complete the following information):**
 - OS [e.g. Ubuntu 18.04, Raspbian]
 OpenWRT 24.10
 - Hardware [e.g. x86, rpi, nanopi neo2 ]
 TP-Link EAP225
 - WiFi cards [e.g. rt28xx, rtl8812au]
 QCA9561
 - WiFi drivers [e.g. stock, patched from http://.. ]

**Additional context**
Removing a single letter and rebuilding the .deb solves this.
``` '/bin/ash', ```

**Confirm you read**
- [ ]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [ ]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

aepkolol, 2025-01-12:
Also, if someone gets here from google about the TP-Link EAP225, I couldn't get injection working on 5ghz. 2.4 was fine, but useless for me. Monitor mode worked flawlessly however. Doesn't seem like a drivers issue since 2.4 and 5 both use the same drivers. Idk. Ordered another type of router to try.

svpcom, 2025-01-13:
@aepkolol Using bash is intended here because there are many bash-only features inside the script. You can install bash to the openwrt router using `opkg update && opkg install bash`

svpcom, 2025-01-13:
@aepkolol  TP-Link EAP225 is ath10k (according to specs on openwrt page). Injection is currently supported by **ath9k** drivers only (rtl8812au/eu have not been seen on any routers). So you need to search quite old models like CPE520, CPE610 (**not 710**) etc. Check that it has **Atheros 9xxx** chipet. Not Qualcomm/Atheros (QCA)

aepkolol, 2025-01-13:
I wasn't aware I could install bash! Sorry, I haven't used openWrt much. 

I spent all day getting openWrt working on an [ASUS AX4200](https://openwrt.org/toh/asus/tuf-ax4200) after it initially bricked the router, but monitor and packet injection works flawlessly on 5ghz with it. It was a last ditch effort to get something I had on hand working, but it paid off. So I can confirm whatever chipset/driver combo is on that router fully works with wfb-ng openWrt. 

I do have a weird issue where as soon as QGC connects to the drone over wfb, the drone stops its mavlink stream though. This happens with every device combo I have tried so far. Telling the GCS to not set vehicle stream rates and not emit heartbeats helps, dunno why. But I only get it with wfb-ng, I need to investigate further. 10 seconds or so after QGC connects, I can see in wfb-cli that no mavlink packets are being sent across from the drone.

svpcom, 2025-01-25:
@aepkolol You got working injection on MT7986AV, MT7976DA chipset? It it possible to control tx power?

svpcom, 2025-01-25:
Could you show output of `wfb-cli gs` and `wfb-cli drone` before and after mavlink failure?