svpcom/wfb-ng #399: [Feature] detect wfb-ng available cards for simple gs installation

lida2003, 2025-01-25:
We usually use `sudo ./scripts/install_gs.sh wlan1` to install gs software.

Currently, I noticed that there is a [auto-detection feature comitted](https://github.com/svpcom/wfb-ng/commit/4610a7110aa0326956a65e43e105092e06cfc631).

And I think we might let `install_gs.sh` script smart enough to detect parameter `wlan1`.

The real issue is ubuntu system is changing the name rule from wlan0/wlan1 etc. There will be an auto-generated name, such as `wlx200db032da3a`

So **detect wfb-ng available cards for simple gs installation** is such kind of feature for people to simple installation with wfb-ng card plug-in when install gs software.

Following consideration might be needed:
- no wfb-ng card in the system: print error for installtion
- only one wfb-ng card: just substitue parameter1
- multi wfb-ng cards
- there are parameters, which seems to be input by force for installtion

Any idea or thoughts, please share here. Thanks!

svpcom, 2025-01-25:
@lida2003 done

lida2003, 2025-01-25:
@svpcom That's really speedy.

Well, it's really automatic, I think very convient for newcomers.

There is another possibility, when people know udev rule/wifi device name very well. And they will install the wfb-ng by scripts and most probable use forced installation without any device connected to the system.

- for newcomers, install wfb with auto detection: `sudo ./scripts/install_gs.sh`
- for people familiar with the wfb-ng(assign a pre-configured wlan device, such as `wfb-rf`) installation without wifi-card plugged: `sudo ./scripts/install_gs.sh wfb-rf` 

Note: wfb-rf will be configured by other scripts, such as udev configuration etc.

BTW, if we want to remove $1 parameter, then readme might be need to be synced and comment about $1 parameter removed. So I think it might be a good choice for older users like me. What do you think?

svpcom, 2025-01-25:
@lida2003 Done

lida2003, 2025-01-26:
I have tested with below changes, please help to check:

```
daniel@daniel-nvidia:~/Work/jetson-fpv/module/wfb-ng$ git diff
diff --git a/scripts/install_gs.sh b/scripts/install_gs.sh
index c85faca..c188557 100755
--- a/scripts/install_gs.sh
+++ b/scripts/install_gs.sh
@@ -38,11 +38,12 @@ mv gs.key /etc/gs.key
 if [ $auto_nics -eq 0 ]
 then
     echo "Saving WFB_NICS=\"$nics\" to /etc/default/wifibroadcast"
-    echo "WFB_NICS=\"$nics\"" > /etc/default/wifibroadcast
 else
-    echo "Using wifi autodetection"
+    echo "Using wifi autodetection WFB_NICS=\"$nics\""
 fi

+echo "WFB_NICS=\"$nics\"" > /etc/default/wifibroadcast
+
 # Setup config
 cat <<EOF >> /etc/wifibroadcast.cfg
 [common]
daniel@daniel-nvidia:~/Work/jetson-fpv/module/wfb-ng$ git log -n 1
commit 33cbb3cd2b12a720eb7a0eb48e8a507c4ece007c (HEAD -> master, origin/master, origin/HEAD)
Author: Vasily Evseenko <svpcom@gmail.com>
Date:   Sun Jan 26 01:41:15 2025 +0300

    Add ability for manual override wifi autodetection #399
```

svpcom, 2025-01-26:
This will disable wifi card autodetection because list of cards will be fixed during setup.
If in the next time user connect different cards then GS will crash

``` .patch
diff --git a/scripts/install_gs.sh b/scripts/install_gs.sh
index c85faca..c188557 100755
--- a/scripts/install_gs.sh
+++ b/scripts/install_gs.sh
@@ -38,11 +38,12 @@ mv gs.key /etc/gs.key
 if [ $auto_nics -eq 0 ]
 then
     echo "Saving WFB_NICS=\"$nics\" to /etc/default/wifibroadcast"
-    echo "WFB_NICS=\"$nics\"" > /etc/default/wifibroadcast
 else
-    echo "Using wifi autodetection"
+    echo "Using wifi autodetection WFB_NICS=\"$nics\""
 fi

+echo "WFB_NICS=\"$nics\"" > /etc/default/wifibroadcast
```

lida2003, 2025-01-27:
With latest code, I rm `/etc/default/wifibroadcast` `/etc/wifibroadcast.cfg` and run `sudo ./scripts/install_gs.sh`.

The gs service will NOT run due to lack of `/etc/default/wifibroadcast`.
And there is no `/etc/default/wifibroadcast` existed if we are using current auto-detection script.
Any thing I missed here?

```
dpkg-genbuildinfo --build=binary -O../wfb-ng_25.1.25.81795-1_arm64.buildinfo
 dpkg-genchanges --build=binary -O../wfb-ng_25.1.25.81795-1_arm64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-source: info: using options from wfb-ng-25.1.25.81795/debian/source/options: --extend-diff-ignore=\.egg-info$
dpkg-buildpackage: info: binary-only upload (no source included)
rm -rf wfb_ng.egg-info/ wfb-ng-25.1.25.81795.tar.gz
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'wfb-ng' instead of './deb_dist/wfb-ng_25.1.25.81795-1_arm64.deb'
wfb-ng is already the newest version (25.1.25.81795-1).
The following packages were automatically installed and are no longer required:
  libavcodec58 libavdevice58 libavfilter7 libavformat58 libavutil56 libpostproc55 libssh-gcrypt-4 libswresample3 libswscale5
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Drone keypair (drone sec + gs pub) saved to drone.key
GS keypair (gs sec + drone pub) saved to gs.key
Using wifi autodetection
Job for wifibroadcast@gs.service failed because of unavailable resources or another system error.
See "systemctl status wifibroadcast@gs.service" and "journalctl -xeu wifibroadcast@gs.service" for details.
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ systemctl status wifibroadcast@gs.service
● wifibroadcast@gs.service - WFB-ng standalone server, profile gs
     Loaded: loaded (/lib/systemd/system/wifibroadcast@.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: resources) since Mon 2025-01-27 08:54:37 CST; 3s ago
        CPU: 0
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ journalctl -xeu wifibroadcast@gs.service
░░ Subject: A stop job for unit wifibroadcast@gs.service has finished
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ A stop job for unit wifibroadcast@gs.service has finished.
░░
░░ The job identifier is 3926 and the job result is done.
1月 27 08:54:47 daniel-nvidia systemd[1]: wifibroadcast@gs.service: Failed to load environment files: No such file or directory
1月 27 08:54:47 daniel-nvidia systemd[1]: wifibroadcast@gs.service: Failed to run 'start' task: No such file or directory
1月 27 08:54:47 daniel-nvidia systemd[1]: wifibroadcast@gs.service: Failed with result 'resources'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ The unit wifibroadcast@gs.service has entered the 'failed' state with result 'resources'.
1月 27 08:54:47 daniel-nvidia systemd[1]: Failed to start WFB-ng standalone server, profile gs.
░░ Subject: A start job for unit wifibroadcast@gs.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░
░░ A start job for unit wifibroadcast@gs.service has finished with a failure.
░░
░░ The job identifier is 3926 and the job result is failed.
```

svpcom, 2025-01-27:
@lida2003 `/etc/default/wifibroadcast` should be a part of wfb-ng deb package. See `setup.py`

lida2003, 2025-01-27:
Yes, as steps as I did:

1. remove configure files: `/etc/default/wifibroadcast`
2. install gs from script: `sudo ./scripts/install_gs.sh`
3. then there will be no `/etc/default/wifibroadcast` file
4. finaly, the service is NOT up and running

> If in the next time user connect different cards then GS will crash

I know what you meant about `difference cards`, but what I understand is smart installation not start autodetection of each app launch, am I right?

svpcom, 2025-01-27:
@lida2003 /etc/default/wifibroadcast is a part of deb package. No need to remove it. By default it has wifi autodetection:
``` .bash
# For multi-link setup you can specify path to custom config
# and/or list of cards in /etc/default/wifibroadcast.<profile_name>
# WIFIBROADCAST_CFG=/etc/wifibroadcast_linkXXX.cfg

# Autodetect local cards supported by wfb-ng (8812au and 8812eu)
WFB_NICS="$(wfb-nics)"

```

lida2003, 2025-01-28:
I remove it and use the script to re-install as the steps above. I think it should be created by default, am I right?

But why setup as I did failed? I'm confused.

svpcom, 2025-01-28:
You need to reinstall deb package. Just do `apt remove wfb-ng` before running install_gs.sh

lida2003, 2025-01-30:
It seems `sudo apt-get remove -y wfb-ng` will not remove `/etc/default/wifibroadcast`.
If manually remove the `/etc/default/wifibroadcast` then it will never start the application due to `/etc/default/wifibroadcast` lost.

```
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ ls /etc/default/wifibroadcast
/etc/default/wifibroadcast
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ cat /etc/default/wifibroadcast
WFB_NICS="wlx200db032da3a"
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ sudo apt-get remove -y wfb-ng
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libavcodec58 libavdevice58 libavfilter7 libavformat58 libavutil56 libpostproc55 libssh-gcrypt-4 libswresample3 libswscale5
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  wfb-ng
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 2,105 kB disk space will be freed.
(Reading database ... 254142 files and directories currently installed.)
Removing wfb-ng (25.1.25.81795-1) ...
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ ls /etc/default/wifibroadcast
/etc/default/wifibroadcast
daniel@daniel-nvidia:~/Work/jetson-fpv/install$ cat /etc/default/wifibroadcast
WFB_NICS="wlx200db032da3a"
```

if I remove this configure file, do as follows, it will never install `/etc/default/wifibroadcast`

```
sudo apt-get remove -y wfb-ng
sudo rm -f /etc/default/wifibroadcast
sudo rm -f /etc/wifibroadcast.cfg

cd ../module/wfb-ng
git submodule update --init --recursive

sudo ./scripts/install_gs.sh
```

svpcom, 2025-01-31:
@lida2003 Also experimental apt repository for wfb-ng is available. You can use it instead of compiling from sources. https://apt.wfb-ng.org/

lida2003, 2025-02-01:
OK, it works, although I have met "dpkg: warning: while removing wfb-ng, directory '/usr/lib/python3/dist-packages/wfb_ng/conf' not empty so not removed". 

But if I remove this folder `/usr/lib/python3/dist-packages/wfb_ng`, then it works. 

```bash
$ sudo apt purge wfb-ng
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libavcodec58 libavdevice58 libavfilter7 libavformat58 libavutil56 libpostproc55 libssh-gcrypt-4 libswresample3 libswscale5
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  wfb-ng*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 254102 files and directories currently installed.)
Purging configuration files for wfb-ng (25.1.25.81795-1) ...
dpkg: warning: while removing wfb-ng, directory '/usr/lib/python3/dist-packages/wfb_ng/conf' not empty so not removed
daniel@daniel-nvidia:~/Work/jetson-fpv$ ls /usr/lib/python3/dist-packages/wfb_ng/conf
__pycache__
$ tree /usr/lib/python3/dist-packages/wfb_ng
/usr/lib/python3/dist-packages/wfb_ng
├── conf
│   └── __pycache__
│       └── __init__.cpython-310.pyc
└── __pycache__
    ├── cluster.cpython-310.pyc
    ├── common.cpython-310.pyc
    ├── config_parser.cpython-310.pyc
    ├── __init__.cpython-310.pyc
    ├── mavlink.cpython-310.pyc
    ├── mavlink_protocol.cpython-310.pyc
    ├── protocols.cpython-310.pyc
    ├── proxy.cpython-310.pyc
    ├── server.cpython-310.pyc
    ├── services.cpython-310.pyc
    └── tuntap.cpython-310.pyc

3 directories, 12 files
daniel@daniel-nvidia:~/Work/jetson-fpv$ sudo rm -rf /usr/lib/python3/dist-packages/wfb_ng
daniel@daniel-nvidia:~/Work/jetson-fpv$ sudo apt purge wfb-ng
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package wfb-ng
daniel@daniel-nvidia:~/Work/jetson-fpv$ sudo rm -f /etc/default/wifibroadcast
daniel@daniel-nvidia:~/Work/jetson-fpv$ sudo rm -f /etc/wifibroadcast.cfg
daniel@daniel-nvidia:~/Work/jetson-fpv/module/wfb-ng$ sudo ./scripts/install_gs.sh
daniel@daniel-nvidia:~/Work/jetson-fpv/module/wfb-ng$ cat /etc/default/wifibroadcast
#WFB_NICS="wlx00c0caa578a9"
#WFB_NICS="wlan1 wlan2"

# For multi-link setup you can specify path to custom config
# and/or list of cards in /etc/default/wifibroadcast.<profile_name>
# WIFIBROADCAST_CFG=/etc/wifibroadcast_linkXXX.cfg

# Autodetect local cards supported by wfb-ng (8812au and 8812eu)
WFB_NICS="$(wfb-nics)"
```

svpcom, 2025-02-01:
@lida2003 "dpkg: warning: while removing wfb-ng, directory '/usr/lib/python3/dist-packages/wfb_ng/conf' not empty so not removed" means that it seems you do some manual edits in master.cfg.

lida2003, 2025-02-01:
Thanks, I DID changed channel in `/etc/wifibroadcast.cfg`. I'll add extra scripts to make sure all configuration will be removed and re-install from start.

closed 2025-02-01 by lida2003