svpcom/wfb-ng #404: WFB-NG Support on Yocto in Docker (Kernel 6.6.54, Verdin iMX8M Plus) - DKMS Build Failing

ArvinVeiyon, 2025-02-09:
Does WFB-NG support running on a Yocto-based OS inside a Docker container?

Board: Toradex Verdin iMX8M Plus (running Yocto-based Toradex BSP).

Kernel: 6.6.54-7.1.0-g3493ccd66900 (aarch64).

Docker Environment: Running inside a Docker container based on Ubuntu 24.04.

Issue: Unable to compile WFB-NG driver inside the Docker container due to missing or incomplete kernel headers.

The same setup works fine on Ubuntu, but inside Yocto Docker, DKMS build fails due to missing headers or symbols.

First Log Indicating Missing Headers:

make[1]: Entering directory '/usr/src/linux-headers-6.6.54-7.1.0-g3493ccd66900'
/usr/src/linux-headers-6.6.54-7.1.0-g3493ccd66900/Makefile:741: include/config/auto.conf: No such file or directory
make[1]: *** [/usr/src/linux-headers-6.6.54-7.1.0-g3493ccd66900/Makefile:784: .config] Error 1
make: *** [Makefile:234: __sub-make] Error 2

Follow-Up Log After Installing Headers inside the docker:

WARNING: Module.symvers is missing.
ERROR: modpost: "skb_pull" [/var/lib/dkms/rtl8812au/5.2.20.2/build/88XXau_wfb.ko] undefined!
ERROR: modpost: "rtnl_is_locked" [/var/lib/dkms/rtl8812au/5.2.20.2/build/88XXau_wfb.ko] undefined!
make[3]: *** [scripts/Makefile.modpost:145: /var/lib/dkms/rtl8812au/5.2.20.2/build/Module.symvers] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.54-7.1.0-g3493ccd66900/Makefile:1865: modpost] Error 2

Question:

Has anyone successfully compiled WFB-NG on Yocto inside Docker?

Are there known issues with DKMS builds for custom Yocto kernels inside Docker?

Any workarounds or suggestions for resolving the missing kernel headers and symbols?

We would really appreciate any insights or guidance on this. Thanks in advance!

svpcom, 2025-02-09:
If you run patched wifi driver outside of container and inject wifi interface inside then it will work, but why do you need run in under the docker?

ArvinVeiyon, 2025-02-10:
Thanks for your suggestion! We haven't tried running the patched WiFi driver outside the container on Yocto OS and injecting the interface inside yet. That sounds like a good approach—we'll check and see how it works.

ArvinVeiyon, 2025-02-11:
We initially tried installing the Realtek RTL WiFi driver inside the Docker container because Torizon OS (Toradex BSP) is based on Yocto and lacks a traditional package manager like apt or opkg, making it difficult to install dependencies directly on the host OS.

 Since Torizon is designed to run applications and drivers in containers, we followed this approach to avoid modifying the base system. However, since this is not ideal, we will now build a new OS image with the patched driver integrated and test it directly on the host. 

Thanks for your response! Please consider this query closed.

closed 2025-02-11 by svpcom