svpcom/wfb-ng #419: Please, Include Missed Packets per Channel in the Statistics

tipoman9, 2025-03-31:
Missed packets per channel are the most important metrics for signal quality.
Much more important then RSSI or SNR.
This is the first sign that the link quality  is going bad for a channel.
Currently wfb_rx exposes these metrics
```
{
    "type": "rx",
    "timestamp": 1743453059.1416976,    
    "packets": {
        "all": [990, 91088],
        "dec_err": [0, 936],
        "dec_ok": [990, 90152],
        "fec_rec": [0, 0],
        "lost": [0, 0],
        "bad": [0, 0],
        "out": [398, 36032],
        "out_bytes": [469674, 42121751]
    },
    "session": {},
    "rx_ant_stats": [{
            "ant": 256,                     
            "pkt_recv": 495,
           ...
          
        }, {
            "ant": 0,          
            "pkt_recv": 495,
			...         
        }
    ]
}
```

Imagine the following scenario.
The Tx sends 500 packets.
Channel 1 receives 495, misses 5 (spread in the first half of the period)
Channel 2 receives 495, misses 5 (spread in the last part of the period)
Since all the missed packets of Channel 1 are received by Channel 2, the aggregator will compensate them and no FEC will be needed, so `fec_rec` will be 0.
And there is no way to tell from the exported statistics that 5 packets were missed by every channel.

I have implemented this metric in an ugly way, so I don't dare to do a PR with such an implementation:
https://github.com/tipoman9/wfb-ng/blob/731c227e5abd096c083723e3ee60420ce9a674db/src/rx.cpp#L498
But I can't imagine flying without it, since digital signal tend to break suddenly, unlike analogue one.
It would be very nice if you implement this feature the proper way in wfb_rx and export it via the TCP statistics.
Here it is how it looks like in flight

[image]

svpcom, 2025-04-02:
@tipoman9 done. See example in wfb-cli

svpcom, 2025-04-04:
@tipoman9 Have you tried feature above?

snokvist, 2025-04-04:
It seems to work! The adapter with "dloss" is my rtl8812bu which i always though was shitty and was losing packets, and this proves it.

[image]

{"type": "rx", "timestamp": 1743787812.9160862, "id": "video rx", "tx_wlan": 2, "packets": {"all": [1209, 503769], "all_bytes": [1300383, 543938136], "dec_err": [0, 490], "session": [3, 1264], "data": [1206, 502015], "uniq": [403, 167886], "fec_rec": [0, 712], "lost": [0, 61], "bad": [0, 0], "out": [268, 112640], "out_bytes": [258914, 109708410]}, "session": {"fec_type": "VDM_RS", "fec_k": 8, "fec_n": 12, "epoch": 0}, "rx_ant_stats": [{"ant": 512, "freq": 5825, "mcs": 0, "bw": 20, "pkt_recv": 403, "rssi_min": -28, "rssi_avg": -16, "rssi_max": -13, "snr_min": -28, "snr_avg": -16, "snr_max": -13}, {"ant": 0, "freq": 5825, "mcs": 0, "bw": 20, "pkt_recv": 403, "rssi_min": -28, "rssi_avg": -16, "rssi_max": -13, "snr_min": -28, "snr_avg": -16, "snr_max": -13}, {"ant": 256, "freq": 5825, "mcs": 0, "bw": 20, "pkt_recv": 400, "rssi_min": -41, "rssi_avg": -38, "rssi_max": -38, "snr_min": 0, "snr_avg": 0, "snr_max": 0}]}

tipoman9, 2025-04-04:
https://github.com/user-attachments/assets/1ac681c0-f52a-4350-bf0c-6c97114713eb

This is how the widget for x86 looks like for three ar9271 adapters based on this statistics.

svpcom, 2025-04-05:
@tipoman9 My only note that nonzero `fec_r` is not only when packet was lost, but also when you have packets reordering