svpcom/wfb-ng #421: [BUG] Unable to install/build rtl8812eu driver on Nvidia Jetson Orin Nano

jmssa, 2025-04-20:
**Describe the bug**
Both "sudo ./dkms-install.sh" and "make" finish with errors when attempting to install rtl8812eu driver.

**To Reproduce**
```
$ git clone -b v5.15.0.1 https://github.com/svpcom/rtl8812eu.git
$ cd rtl8812eu/
$ sudo ./dkms-install.sh  #or make
```

**Expected behavior**
Finishing installation or building and detecting the wifi module the same as my x86 laptop.

**Screenshots**
Not applicable, I believe.

**Your setup (please complete the following information):**
 - OS: Jetpack 6.2 (Based of Ubuntu 22.04)
 - Hardware: Jetson Orin Nano 4gb
 - WiFi cards: LB-Link BL-M8812EU2
 - WiFi drivers: Cloned directly from [https://github.com/svpcom/rtl8812eu.git](https://github.com/svpcom/rtl8812eu.git)

**Additional context**
The make.log results in

```
DKMS make.log for rtl88x2eu-5.15.0.1 for kernel 5.15.148-tegra (aarch64)
Sun Apr 20 08:02:11 PM WEST 2025
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.148-tegra/build M=/var/lib/dkms/rtl88x2eu/5.15.0.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.148-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_regdb_rtk.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_chset.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_dfs.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_txpwr.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_p2p.o
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_recv.c: In function ‘validate_mp_recv_frame’:
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_recv.c:3906:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3906 |                         for (i = 0; i < precv_frame->u.hdr.len; i = i + 8)
      |                         ^~~
In file included from /var/lib/dkms/rtl88x2eu/5.15.0.1/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_recv.c:17:
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/core/rtw_recv.c:3909:33: note: in expansion of macro ‘RTW_INFO’
 3909 |                                 RTW_INFO("#############################\n");
[... 105 lines cut ...]
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.c:11065:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
11065 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.c:11065:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
  CC [M]  /var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/rtw_radiotap.o
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.c:11330:17: error: ‘struct wireless_dev’ has no member named ‘valid_links’
11330 |         if (wdev->valid_links && wdev->links[0].client.current_bss)
      |                 ^~
/var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.c:11330:38: error: ‘struct wireless_dev’ has no member named ‘links’
11330 |         if (wdev->valid_links && wdev->links[0].client.current_bss)
      |                                      ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:295: /var/lib/dkms/rtl88x2eu/5.15.0.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1912: /var/lib/dkms/rtl88x2eu/5.15.0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.148-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
make: *** [Makefile:2600: modules] Error 2
```

**Confirm you read**
- [x]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [x]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

Thanks in advance for any help!

aepkolol, 2025-04-24:
I had this issue a while ago. I think the issue was jetpack does some weird kernel mods to cfg80211. When you go to compile the drivers, it has checks for what kernel version you have an adjusts some variables based on that. However the kernel patches nvidia added makes those checks useless. I spent way too long fixing that up to get it to compile.

I added the pre-compiled driver to this git https://github.com/aepkolol/jetson-8812eu

jmssa, 2025-04-29:
Thank you very much for your reply. Which version of Jetpack are you using? I tried to load the module with modprobe and got 
`modprobe: ERROR: could not insert '8812eu_jetson': Exec format error`
Which I believe may be caused by the kernel version in which the driver was compiled being different from the one I am currently running, since dmseg outputs
`8812eu: disagrees about version of symbol module_layout`

It should be worth mentioning that I downgraded to Jetpack 6.1 since the original post for other reasons relating to pytorch.