svpcom/wfb-ng #428: [BUG] can't decode video strem using gstremer

krzysztofkuczek, 2025-06-03:
**Describe the bug**
There is no way to decode h264 stream using proposed gstreamer pipeline.
The video is working fine using qgroundcontrol

The version of the depay plugin is like below
```

qczek@kubuntu-t490:~$ gst-inspect-1.0 rtph264depay
Factory Details:
  Rank                     secondary (128)
  Long-name                RTP H264 depayloader
  Klass                    Codec/Depayloader/Network/RTP
  Description              Extracts H264 video from RTP packets (RFC 3984)
  Author                   Wim Taymans <wim.taymans@gmail.com>
  Documentation            https://gstreamer.freedesktop.org/documentation/rtp/rtph264depay.html

Plugin Details:
  Name                     rtp
  Description              Real-time protocol plugins
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtp.so
  Version                  1.24.2
  License                  LGPL
  Source module            gst-plugins-good
  Documentation            https://gstreamer.freedesktop.org/documentation/rtp/
  Source release date      2024-04-09
  Binary package           GStreamer Good Plugins (Ubuntu)
  Origin URL               https://launchpad.net/ubuntu/+source/gst-plugins-good1.0

```

```
qczek@kubuntu-t490:~$ gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
               ! rtph264depay ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'autovideosink0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayGBM\)\ gldisplaygbm0";
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
WARNING: from element /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0: Could not decode stream.
Additional debug info:
../gst/rtp/gstrtph264depay.c(1453): gst_rtp_h264_depay_process (): /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0:
Undefined packet type
WARNING: from element /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0: Could not decode stream.
Additional debug info:
../gst/rtp/gstrtph264depay.c(1453): gst_rtp_h264_depay_process (): /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0:
Undefined packet type
WARNING: from element /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0: Could not decode stream.
Additional debug info:
../gst/rtp/gstrtph264depay.c(1453): gst_rtp_h264_depay_process (): /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0:
Undefined packet type
^Chandling interrupt.
```

**To Reproduce**
Open the console and run:
gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
               ! rtph264depay ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink sync=false

**Expected behavior**
Video output will be visible, there will be no errors 

**Screenshots**
N/A

**Your setup (please complete the following information):**
 - OSLinux kubuntu-t490 6.11.0-25-generic #25~24.04.1-Ubuntu 
 - Hardware x86
 - WiFi cards N/A the steam comes from another board running wfb-ng
 - WiFi drivers N/A the steam comes from another board running wfb-ng

**Additional context**
N/A

**Confirm you read**
- [ X]  https://github.com/svpcom/wfb-ng/blob/master/README.md
- [ X]  https://github.com/svpcom/wfb-ng/wiki/Setup-HOWTO

closed 2025-06-03 by krzysztofkuczek

svpcom, 2025-06-03:
You miss `h264parse` element it pipeline above

krzysztofkuczek, 2025-06-03:
thank you. The problem was with h264 used instead h265....

h4ppyr0gu3, 2026-01-28:
please could you expand @svpcom I have tried to decode the stream adding h264parse after rtph264depay but it is actually failing at rtph264depay, before that i get the following error
```
 gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
              ! h264parse ! rtph264depay ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink sync=false
WARNING: erroneous pipeline: could not link udpsrc0 to h264parse0

```

if i then run it with h264parse before i end up with:
```
gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
              ! rtph264depay ! h264parse ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
WARNING: from element /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0: Could not decode stream.
Additional debug info:
../gst/rtp/gstrtph264depay.c(1453): gst_rtp_h264_depay_process (): /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0:
Undefined packet type
WARNING: from element /GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0: Could not decode stream.
```